ARMmbed/mbed-cli #300: Certificate Verify Failed - hg repos

BlackstoneEngineering, 2016-08-01:
## Problem
on a vanilla machine hg clone doesnt work. 

I try running `mbed import https://developer.mbed.org/teams/mbed/code/mbed_blinky/` and get the following 
```shell

C:\Users\ARM>hg clone https://developer.mbed.org/teams/mbed/code/mbed_blinky/
abort: error: [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed (_ssl.c
:590)

C:\Users\ARM>mbed import https://developer.mbed.org/teams/mbed/code/mbed_blinky/

[mbed] Working path "C:\Users\ARM" (directory)
[mbed] Cloning C:\Users\ARM\mbed_blinky from https://mbed.org/teams/mbed/code/mb
ed_blinky/
[mbed] "hg clone https://mbed.org/teams/mbed/code/mbed_blinky/ C:\Users\ARM\mbed
_blinky"
abort: error: [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed (_ssl.c
:590)
[mbed] Cloning C:\Users\ARM\mbed_blinky from https://mbed.org/teams/mbed/code/mb
ed_blinky/
[mbed] "git clone https://mbed.org/teams/mbed/code/mbed_blinky/ C:\Users\ARM\mbe
d_blinky"
Traceback (most recent call last):
  File "C:\Python27\Scripts\mbed-script.py", line 9, in <module>
    load_entry_point('mbed-cli==0.8.9', 'console_scripts', 'mbed')()
  File "C:\Python27\lib\site-packages\pkg_resources\__init__.py", line 542, in l
oad_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "C:\Python27\lib\site-packages\pkg_resources\__init__.py", line 2569, in
load_entry_point
    return ep.load()
  File "C:\Python27\lib\site-packages\pkg_resources\__init__.py", line 2229, in
load
    return self.resolve()
  File "C:\Python27\lib\site-packages\pkg_resources\__init__.py", line 2235, in
resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "build\bdist.win32\egg\mbed\mbed.py", line 1153, in <module>
  File "build\bdist.win32\egg\mbed\mbed.py", line 723, in thunk
  File "build\bdist.win32\egg\mbed\mbed.py", line 780, in import_
  File "build\bdist.win32\egg\mbed\mbed.py", line 357, in clone
  File "build\bdist.win32\egg\mbed\mbed.py", line 113, in popen
  File "C:\Python27\lib\subprocess.py", line 710, in __init__
    errread, errwrite)
  File "C:\Python27\lib\subprocess.py", line 958, in _execute_child
    startupinfo)
WindowsError: [Error 2] The system cannot find the file specified
```

## Solution
Check if `hg clone` fails, then prompt the user to use `hg clone --insecure`

Something like this

```
$ mbed import example.com/reponame
......
[mbed] Repo failed to import due to invalid security certs. See output above. Would you like to try an insecure impor? [Y/n]
$ Y
.........
[mbed] Success!
```

screamerbg, 2016-08-01:
@BlackstoneEngineering perhaps the easiest would be to fix the certificate issues? Happy to report this to the web team?

BlackstoneEngineering, 2016-08-01:
not on the web team, this is a windows glitch, happens when someone doesn't update their machine for a long period of time. In the case of my VM ~1 year

screamerbg, 2016-08-03:
@BlackstoneEngineering how can I trigger/reproduce the certificate error?

BlackstoneEngineering, 2016-08-03:
not entirely sure tbh, havent managed to solve it either, so not sure how to root cause it, just know its a thing, more work tbd

BlackstoneEngineering, 2016-08-03:
not sure, leave it open so when others run into it (which they will, this is a thing) they can chime in here and we can track it?

screamerbg, 2016-08-04:
Now part of mbed CLI 0.9.1 release

closed 2016-08-04 by screamerbg