ARMmbed/mbed-cli #301: pretty table need to be added to mbed-cli requirements.txt

BlackstoneEngineering, 2016-08-01:
## Problem
On a clean machine I installed mercurial, python and gcc, then I did `pip install mbed-cli`. Next I did `mbed import https://developer.mbed.org/teams/mbed/code/mbed_blinky/` followed by `mbed deploy`. I then said `mbed config toolchain GCC_ARM` and `mbed config target K64F`. Finally I went to compile the program with `mbed compile`, and which point the compile promptly terminated because I didnt have `prettytable` installed. 

See output here 
```shell
C:\Users\ARM\mbed_blinky>mbed compile
Traceback (most recent call last):
  File "C:\Users\ARM\mbed_blinky\.temp\tools\make.py", line 43, in <module>
    from tools.options import get_default_options_parser
  File "C:\Users\ARM\mbed_blinky\.temp\tools\options.py", line 18, in <module>
    from tools.toolchains import TOOLCHAINS
  File "C:\Users\ARM\mbed_blinky\.temp\tools\toolchains\__init__.py", line 34, i
n <module>
    from tools.memap import MemapParser
  File "C:\Users\ARM\mbed_blinky\.temp\tools\memap.py", line 14, in <module>
    from prettytable import PrettyTable
ImportError: No module named prettytable
[mbed] ERROR: "python" returned error code 1.
[mbed] ERROR: Command "python -u C:\Users\ARM\mbed_blinky\.temp\tools\make.py -t
 GCC_ARM -m K64F --source . --build .\.build\K64F\GCC_ARM" in "C:\Users\ARM\mbed
_blinky"
---
```

## Solution
Either add pretty table to the requirements for mbed-cli, or do some magic when importing the mbed 2.0 programs and associated tools. I dont care, i just want the damn thing to work.

BlackstoneEngineering, 2016-08-01:
@screamerbg

BlackstoneEngineering, 2016-08-01:
To get this thing to work i've now had to install jinja2 and colorama aswell as prettytable manually. If there is a fix that already exists for this please push it public and mainline as this inability to have tools that work is a critical blocker to making the documentation videos for mbed-cli

screamerbg, 2016-08-01:
@BlackstoneEngineering or for this release the guides can point to `pip install -r requirements.txt`. https://github.com/ARMmbed/mbed-os/pull/2321 was reverted thus for now the tools won't be able to support automatic handling of dependencies

BlackstoneEngineering, 2016-08-01:
dude, this isnt for mbedOS, this is for mbed classic

BlackstoneEngineering, 2016-08-01:
your tool doesnt work properly for mbed classic

screamerbg, 2016-08-03:
The requirements.txt has been added to the 2.0 tools and now missing requirements are detected when mbed 2.0 program is imported, e.g.

```
mbed import https://developer.mbed.org/users/samux/code/USBSerial_HelloWorld usb-hellowold
[mbed] Importing program "usb-hellowold" from "https://developer.mbed.org/users/samux/code/USBSerial_HelloWorld" at latest revision in the current branch
[mbed] Adding library "mbed" from "http://mbed.org/users/mbed_official/code/mbed/builds" at rev #3d0ef94e36ec
[mbed] Downloading mbed library build "3d0ef94e36ec" (might take a minute)
[mbed] Unpacking mbed library build "3d0ef94e36ec" in "D:\Work\examples\usb-hellowold\mbed"
[mbed] Adding library "USBDevice" from "https://mbed.org/users/mbed_official/code/USBDevice" at rev #335f2506f422
[mbed] Couldn't find build tools in your program. Downloading the mbed 2.0 SDK tools...
[mbed] WARNING: -----------------------------------------------------------------
[mbed] WARNING: The mbed build tools in this program require Python modules that are not installed.
[mbed] WARNING: This might prevent compiling code or exporting to IDEs and other toolchains.
[mbed] WARNING: The missing Python modules are: mbed-greentea
[mbed] WARNING: You can install all missing modules by running "pip install -r requirements.txt" in "D:\Work\examples\usb-hellowold\.temp\tools"
```

closed 2016-08-03 by screamerbg