ARMmbed/mbed-cli #302: config - need way to list config

BlackstoneEngineering, 2016-08-01:
## Problem
There is no way to know the current config for a repo, you must know what keys to type in to get their values. There needs to be a command to let you see all config for a repo at a glance.

## Solution
Add a `-l` or '-ls` option to `config` so you can do something like this

```
$mbed config toolchain tool
$mbed config target targ
$mbed config -G protocol ssh
.....

$mbed config -ls
[mbed] Configuration is:
  Local
    toolchain : tool
    target : targ
  Global
    protocol : ssh
```

This would implicitly only show globals if they are not explicitly overridden in the local context.

bogdanm, 2016-08-01:
You can do `mbed compile --config`. A `mbed config` command used to exist, but was removed for reasons that I personally never understood.

BlackstoneEngineering, 2016-08-01:
right, i'm asking, as US lead for Developer enablement, to put it back. Less as a engineering thing and more as a story thing. if we can set it we need to be able to validate that we set it correctly. This will prevent endless headaches in workshops and trainings.

screamerbg, 2016-08-03:
Example:

```
[mbed] Global config:
ARM_PATH=C:\Program Files\ARM
CACHE=D:\Temp\repo-cache
TOOLCHAIN=ARM
TARGET=K64F

[mbed] Local config (D:\Work\examples\mbed-os-example-ble-BatteryLevel):
PROTOCOL=ssh
```

bridadan, 2016-08-03:
@bogdanm Pretty sure @BlackstoneEngineering is talking about mbed CLI config, not the mbed OS config system values.

bogdanm, 2016-08-03:
I believe you are completely right :grin:

screamerbg, 2016-08-04:
Now part of mbed CLI 0.9.1 release

closed 2016-08-04 by screamerbg