ARMmbed/mbed-cli #303: requirements.txt - make experience more fluid

BlackstoneEngineering, 2016-08-02:
## Problem
When a user clones a repo, the tools have their own requirements.txt, this requirements.txt may contain things that are not currently installed. Right now the mbed-cli tool spits out a warning, but its not easy to understand or to use. 

The warning looks like this: 
```
[mbed] Couldn't find build tools in your program. Downloading the mbed 2.0 SDK tools...
[mbed] WARNING: -----------------------------------------------------------------
[mbed] WARNING: The mbed build tools in this program require Python modules that are not installed.
[mbed] WARNING: This might prevent compiling code or exporting to IDEs and other toolchains.
[mbed] WARNING: The missing Python modules are: pyserial, intelhex, project-generator, project-generator-definitions, junit-xml, pyyaml, requests, mbe
d-ls, mbed-host-tests, mbed-greentea
[mbed] WARNING: You can install all missing modules by running "pip install -r requirements.txt" in "C:\Users\ARM\scratch\.temp\tools"
---
```
which then requires thew user to manually type the following
```
> pip install -r C:\Users\ARM\scratch\.temp\tools
```

## Solution
This experience is not as easy as it should be. There are many ways we could solve this. From best to worst here they are

### Best Solution
mbed cli detects the tools need additional packages and automatically installs them for the user. Everything is transparent to the end user.

#### Best Solution: Considerations
 I understand that this may violate some peoples idea of what a tool should, but I dont care, mbed-cli needs to be the easiest, best way to do the thing. To appease these pedants we can enable / disable this auto-installation process via a global config flag, which will be turned on by default, thus ensuring the best experience for the majority of users while permitting corner cases to have their cake and eat it too. 

### Reasonable Solution
mbed-cli detects the tools need additional packages and asks the user if they want to install them. This would look like this:
```
[mbed] WARNING: -----------------------------------------------------------------
[mbed] WARNING: The mbed build tools in this program require Python modules that are not installed.
[mbed] WARNING: This might prevent compiling code or exporting to IDEs and other toolchains.
[mbed] WARNING: The missing Python modules are: pyserial, intelhex, project-generator, project-generator-definitions, junit-xml, pyyaml, requests, mbe
d-ls, mbed-host-tests, mbed-greentea
[mbed] WARNING: Would you like mbed-cli to install the missing packages? [Y/n] : 

> Y

[mbed] Installing packages now ........
......
[mbed] Packages ........ were installed succesfully!
```
This is an acceptable compromise that I think we can get out the door before release. I reccomend doing this for now and the best solution later. 

### Acceptable Solution
mbed-cli detects missing packages, and spits out the command for the user to copy/paste.
This is the lazyest way to do this, it interrupts the work flow, but at least it is considerably better than the current solution. 

```
[mbed] WARNING: -----------------------------------------------------------------
[mbed] WARNING: The mbed build tools in this program require Python modules that are not installed
....
....
[mbed] WARNING: You can install all missing modules by running "pip install -r C:\Users\ARM\scratch\.temp\tools\requirements.txt"
```
At the bare minimum we should make the command easy to assemble to copy/paste for the end user.

screamerbg, 2016-08-03:
@sg- @theotherjimmy

screamerbg, 2016-08-03:
These mbed 2 and mbed OS 5 are de-facto the same tools, except that the mbed 2.0 are delivered separately and also that they support compatibility/legacy compile flags for both ARMCC and GCC. Once the automatic handling of mbed OS 5 tools requirements is resolved, this will be propagated to mbed 2 as well.

screamerbg, 2016-08-03:
Introduced in PR #310

screamerbg, 2016-08-04:
Now part of mbed CLI 0.9.1 release

closed 2016-08-04 by screamerbg