ARMmbed/mbed-cli #306: config toolchain - need tab completion to list available options

BlackstoneEngineering, 2016-08-02:
## Problem
nowhere in the mbed-cli help menus does it list what the acceptable options are for `mbed config toolchain`

## Solution
1. add a complete list to the help flavor text for toolchain
2. add tab completion and allow user to cycle thorugh list with tab completion

BlackstoneEngineering, 2016-08-02:
I think @theotherjimmy has some pending PR's that would make tab completion rediculously simple to add.

theotherjimmy, 2016-08-02:
Well, no. It's nowhere near ridiculously simple.

screamerbg, 2016-08-03:
Tab completion is a duplicate of https://github.com/ARMmbed/mbed-cli/issues/241. Please read @mjs-arm and @bremoran's comments.

BlackstoneEngineering, 2016-08-03:
Oi, dont close this, this is a super valid bug! This bug isnt about addint tab completion, its about adding a way to easily specify the toolchain. The other ticket is about git, which is wrapping another tool, this is about our own provided options.

screamerbg, 2016-08-03:
```
$ mbed toolchain -h
usage: mbed toolchain [-h] [-G] [-S] [-v] [-vv] [name]

Set or get default toolchain
This is an alias to 'mbed config [--global] toolchain [name]'

positional arguments:
  name                 Default toolchain name. Example: ARM, uARM, GCC_ARM, IAR
```

theotherjimmy, 2017-11-28:
Closing, as "a way to easily specify toolchain" is already present, as shown above.

closed 2017-11-28 by theotherjimmy