ARMmbed/mbed-cli #308: mbed config target - need ability to figure out whats plugged in

BlackstoneEngineering, 2016-08-03:
## Problem
There is no fluid way of setting the target based on what is plugged into your system. The best path right now is to run `mbed detect` and then copy / paste the target name into `mbed config target <>`. This needs to be easier, idiot proof.

## Solution
Some ideas
1. `mbed detect

screamerbg, 2016-10-19:
@BlackstoneEngineering There is a PR open regarding this feature https://github.com/ARMmbed/mbed-cli/pull/375

Basically you can `mbed compile -t ARM -m detect` or even

```
$ mbed config -G target detect
$ mbed compile -t ARM # target will be always populated based on the detected platform
```

BlackstoneEngineering, 2016-10-19:
So will this work?
Assume I have no toolchain or target set, and a board plugged in. 
```
mbed toolchain GCC_ARM
mbed compile
```

Will this autodetect the board and use it as a target?

bridadan, 2016-10-19:
@BlackstoneEngineering Pretty sure that PR won't set `auto` as the default target. You'd still need to explicitly run `mbed target auto`. @screamerbg Maybe that should be added? Seems like a fairly sane default.

screamerbg, 2016-10-19:
@BlackstoneEngineering Yes, that will work assuming that you set the target to be autodetect. E.g.
```
$ mbed toolchain GCC_ARM
$ mbed target detect # 'auto' can work too
$ mbed compile
```

@bridadan For now it's not a default option.

BlackstoneEngineering, 2016-10-19:
@screamerbg ah, gotcha, so `auto` is a special target name that will default to the first board enumerated by mbedls ?

@bridadan I agree that auto should be the default option. But this is a bigger conversation about the tools story and expected workflows. Ideally I'd like us to explicitly call out 2 different workflows,
1. for CI / compiler backend - can be enabled by `mbed config mode CI` - provides defaults for CI / Online compiler workflows, effectively a super pedantic mode that assumes nothing
2. for end developers - this is the default and makes handy assumptions. 
  * detects toolchains / targets automaticall
  * detects SSH Certs if you have them and uses them
  * other stuff that I cant think of right now

screamerbg, 2016-10-27:
> But this is a bigger conversation about the tools story and expected workflows. Ideally I'd like us to explicitly call out 2 different workflows...

Agreed @BlackstoneEngineering. This is now merged to master.

screamerbg, 2017-07-05:
@BlackstoneEngineering This is now part of mbed CLI 1.1/1.2. Thanks for the contribution/suggestion!

closed 2017-07-05 by screamerbg