ARMmbed/mbed-cli #312: .mbed, .mbed-ls and similar folders should be hidden on windows

BlackstoneEngineering, 2016-08-04:
## Problem
mbed-cli is generating clutter in my explorer view. If I accidentally run mbed-ls in a folder that doesnt have tools it will fail, but not before generating .mbed and .mbed-ls folders in the directory. More over these folders are not hidden and thus generate lots of visual clutter.

## Solution
1. If no tools are found and the command errors remove the temporary directory
2. make .mbed, .mbed-ls and similar folders hidden on windows

theotherjimmy, 2017-11-28:
`.mbed` is the only file managed by Mbed CLI.

screamerbg, 2018-03-28:
There's the counter argument where users want to have these files visible (e.g. "I'm on Linux and the .mbed folder is hidden for me"). I'd rather leave this to the user to decide.

Will close the issue but feel free to reopen if needed.

closed 2018-03-28 by screamerbg