ARMmbed/mbed-cli #313: using --source for header files

mottigondabi, 2016-08-04:
Compiling a library by using a full path-to-file for each c / cpp / h result in a "**some-header-file.h**: No such file or directory" error although **some-header-file.h** exists in the `--source` list.

i.e.:
`mbed compile --library --artifact-name oma -t GCC_ARM -m K64F -DTARGET_LIKE_FRDM_K64F_GCC -o debug-info -DDEBUG -DSA_PV_LOG_LEVEL=SA_PV_LOG_LEVEL_INFO -DSA_PV_PLAT_K64F_MBEDOS_GNUC -v --source /sharedhome/motgon01/provisioning/lib_common/spv_lib_status.h --source /sharedhome/motgon01/provisioning/lib_common/spv_lib_common_defs.h --source /sharedhome/motgon01/provisioning/lib_iot/spv_lib_iot.h --source /sharedhome/motgon01/provisioning/lib_iot/spv_lib_iot_doxygen.h --source /sharedhome/motgon01/provisioning/lib_iot/spv_lib_iot_test_only.h --source /sharedhome/motgon01/provisioning/common/lib_tester_ta_4tzinfra.h --source /sharedhome/motgon01/provisioning/common/mbed_file_utils.h --source /sharedhome/motgon01/provisioning/common/pv_asset_common.h --source /sharedhome/motgon01/provisioning/common/pv_base64.h --source /sharedhome/motgon01/provisioning/common/pv_bigint_utils.h --source /sharedhome/motgon01/provisioning/common/pv_build_info.h --source /sharedhome/motgon01/provisioning/common/pv_build_info_test_helpers.h --source /sharedhome/motgon01/provisioning/common/pv_common_utils.h --source /sharedhome/motgon01/provisioning/common/pv_defs.h --source /sharedhome/motgon01/provisioning/common/pv_endian.h --source /sharedhome/motgon01/provisioning/common/pv_error_handling.h --source /sharedhome/motgon01/provisioning/common/pv_link_pb.h --source /sharedhome/motgon01/provisioning/common/pv_log.h --source /sharedhome/motgon01/provisioning/common/pv_lsocket_client.h --source /sharedhome/motgon01/provisioning/common/pv_lsocket_common.h --source /sharedhome/motgon01/provisioning/common/pv_lsocket_listener.h --source /sharedhome/motgon01/provisioning/common/pv_macros.h --source /sharedhome/motgon01/provisioning/common/pv_mailbox.h --source /sharedhome/motgon01/provisioning/common/pv_message_pb.h --source /sharedhome/motgon01/provisioning/common/pv_pb.h --source /sharedhome/motgon01/provisioning/common/pv_protobuf.h --source /sharedhome/motgon01/provisioning/common/pv_prs_common.h --source /sharedhome/motgon01/provisioning/common/pv_button_handler.h --source /sharedhome/motgon01/provisioning/common/pv_common_msg.h --source /sharedhome/motgon01/provisioning/common/pv_timer.h --source /sharedhome/motgon01/provisioning/common/pv_tunnel_pb.h --source /sharedhome/motgon01/provisioning/common/pv_utils.h --source /sharedhome/motgon01/provisioning/common/pv_wrapping.h --source /sharedhome/motgon01/provisioning/common/pv_pbkdf2.h --source /sharedhome/motgon01/provisioning/common/pv_uvisor_box_defs.h --source /sharedhome/motgon01/provisioning/common/pv_test_common_helpers.h --source /sharedhome/motgon01/provisioning/porting_layer/utils/include/pv_encap_utils.h --source /sharedhome/motgon01/provisioning/porting_layer/utils/stdc/int_include/pv_ctype_utils.h --source /sharedhome/motgon01/provisioning/porting_layer/utils/stdc/int_include/pv_mem_utils.h --source /sharedhome/motgon01/provisioning/porting_layer/utils/stdc/int_include/pv_string_utils.h --source /sharedhome/motgon01/provisioning/out/gen/build_info/pv_build_info_defs.h --source /sharedhome/motgon01/provisioning/out/gen/build_info/factory_build_info_defs.h --source /sharedhome/motgon01/provisioning/out/gen/build_info/identity_build_info_defs.h --source /sharedhome/motgon01/provisioning/out/gen/build_info/oma_client_build_info_defs.h --source /sharedhome/motgon01/provisioning/out/gen/build_info/ftd_build_info_defs.h --source /sharedhome/motgon01/provisioning/out/gen/constants/spv_lib_constants.h --source /sharedhome/motgon01/provisioning/out/gen/constants/pv_constants.h --source /sharedhome/motgon01/provisioning/out/gen/constants/factory_lib_constants.h --source /sharedhome/motgon01/provisioning/out/gen/constants/identity_lib_constants.h --source /sharedhome/motgon01/provisioning/out/gen/constants/oma_client_lib_constants.h --source /sharedhome/motgon01/provisioning/out/gen/constants/ftd_constants.h --source /sharedhome/motgon01/provisioning/services/oma_client/oma_client.h --source /sharedhome/motgon01/provisioning/services/oma_client/oma_client_build_info.h --source /sharedhome/motgon01/provisioning/services/oma_client/spv_oma_client.h --source /sharedhome/motgon01/provisioning/services/oma_client/spv_oma_client_doxygen.h --source /sharedhome/motgon01/provisioning/services/oma_client/oma_client.cpp --source /sharedhome/motgon01/provisioning/services/oma_client/spv_oma_client.cpp --source /sharedhome/motgon01/provisioning/services/oma_client/oma_client_build_info.c --source .`

result in the following error:

`emp/./oma_client.o /sharedhome/motgon01/provisioning/services/oma_client/oma_client.cpp
[DEBUG] Return: 1
[DEBUG] Output: In file included from /sharedhome/motgon01/provisioning/services/oma_client/oma_client.h:18:0,
[DEBUG] Output:                  from /sharedhome/motgon01/provisioning/services/oma_client/oma_client.cpp:18:
[DEBUG] Output: /sharedhome/motgon01/provisioning/services/oma_client/spv_oma_client.h:19:33: fatal error: spv_lib_common_defs.h: No such file or directory
[DEBUG] Output:  #include "spv_lib_common_defs.h"
[DEBUG] Output:                                  ^
[DEBUG] Output: compilation terminated.
Traceback (most recent call last):
  File "/sharedhome/motgon01/provisioning/out/k64f_mbedos_gnuc-debug/pv-client/mbed-os/core/tools/build.py", line 248, in <module>
    name=options.artifact_name)
  File "/sharedhome/motgon01/provisioning/out/k64f_mbedos_gnuc-debug/pv-client/mbed-os/core/tools/build_api.py", line 410, in build_library
    raise e
ToolException: In file included from /sharedhome/motgon01/provisioning/services/oma_client/oma_client.h:18:0,
                 from /sharedhome/motgon01/provisioning/services/oma_client/oma_client.cpp:18:
/sharedhome/motgon01/provisioning/services/oma_client/spv_oma_client.h:19:33: **fatal error: spv_lib_common_defs.h**: No such file or directory
 #include "spv_lib_common_defs.h"
                                 ^
compilation terminated.`

Although **spv_lib_common_defs.h** is on the list.

@screamerbg 
@bogdanm 
@jenia81

screamerbg, 2016-08-04:
@mottigondabi --source is intended for source files only (.c/.cpp/.s), hence the name "--source". Also could you please report this to the https://github.com/ARMmbed/mbed-os repository?

screamerbg, 2016-09-12:
Duplicate of https://github.com/ARMmbed/mbed-os/issues/2326

closed 2016-09-12 by screamerbg