ARMmbed/mbed-cli #321: Report unsupported target before attempting to run compile command

zgoda, 2016-08-08:
```
$ mbed compile -t GCC_ARM -m BLUEPILL_F103C8
Building project mt01 (BLUEPILL_F103C8, GCC_ARM)
Scan: .
Scan: mbed
Scan: env
Compile: main.cpp
[ERROR] In file included from ./mbed/f9eeca106725/mbed.h:22:0,
                 from ./main.cpp:1:
./mbed/f9eeca106725/platform.h:21:20: fatal error: device.h: No such file or directory
 #include "device.h"
                    ^
compilation terminated.

[mbed] ERROR: "python" returned error code 1.
[mbed] ERROR: Command "python -u /home/jazg/tmp/mt01/.temp/tools/make.py -t GCC_ARM -m BLUEPILL_F103C8 --source . --build ./.build/BLUEPILL_F103C8/GCC_ARM" in "/home/jazg/tmp/mt01"
```

The board is not supported by mbed-os and mbed-cli, target is not exported in `./mbed/f9eeca106725`, but `-S` option to `compile` command show it in the matrix.

bridadan, 2016-08-08:
@zgoda Thanks for reporting the issue. I tried running `mbed compile -S` from the `mbed-os-5.1.0` release (commit 0712b8adf6bbc7eb796d5dac26f95d79d40745ef) and I didn't see `BLUEPILL_F103C8` in the matrix. Could you please post the output of the matrix you're seeing and what commit you're using?

zgoda, 2016-08-08:
I was following [mbed-cli handbook](https://docs.mbed.com/docs/mbed-os-handbook/en/5.1/dev_tools/cli/#creating-and-importing-programs).

Okay, first things first:

```
$ mbed --version
0.9.1
```

mbed-cli is installed in virtualenv. Now:

```
$ mbed new mt01 --mbedlib
[mbed] Creating new program "mt01" (git)
[mbed] Adding library "mbed" from "https://mbed.org/users/mbed_official/code/mbed/builds" at latest revision in the current branch
[mbed] Downloading mbed library build "f9eeca106725" (might take a minute)
[mbed] Unpacking mbed library build "f9eeca106725" in "/home/jazg/tmp/mt01/mbed"
[mbed] Updating reference "mbed" -> "https://mbed.org/users/mbed_official/code/mbed/builds/f9eeca106725"
[mbed] Couldn't find build tools in your program. Downloading the mbed 2.0 SDK tools...
$ ls -la mt01
razem 256
drwxrwxr-x  5 jazg jazg   4096 sie  8 17:16 .
drwxrwxr-x 20 jazg jazg 204800 sie  8 17:13 ..
drwxrwxr-x  7 jazg jazg   4096 sie  8 17:16 .git
drwxrwxr-x  4 jazg jazg   4096 sie  8 17:16 mbed
-rw-rw-r--  1 jazg jazg      7 sie  8 17:13 .mbed
-rw-rw-r--  1 jazg jazg     67 sie  8 17:16 mbed.bld
-rw-rw-r--  1 jazg jazg   1329 sie  8 17:16 mbed_settings.py
drwxrwxr-x  3 jazg jazg   4096 sie  8 17:16 .temp
```

Add some code to `main.cpp` and try to compile:

```
$ mbed compile -t GCC_ARM -m BLUEPILL_F103C8
Building project mt01 (BLUEPILL_F103C8, GCC_ARM)
Scan: .
Scan: mbed
Scan: env
Compile: main.cpp
[ERROR] In file included from ./mbed/f9eeca106725/mbed.h:22:0,
                 from ./main.cpp:1:
./mbed/f9eeca106725/platform.h:21:20: fatal error: device.h: No such file or directory
 #include "device.h"
                    ^
compilation terminated.

[mbed] ERROR: "python" returned error code 1.
[mbed] ERROR: Command "python -u /home/jazg/tmp/mt01/.temp/tools/make.py -t GCC_ARM -m BLUEPILL_F103C8 --source . --build ./.build/BLUEPILL_F103C8/GCC_ARM" in "/home/jazg/tmp/mt01"
---
```

Fine, did not expect anything else, this board is not supported by mbed-os. But in the same directory:

```
$ mbed compile -S | grep BLUEPILL
| BLUEPILL_F103C8       |     -     |     -     |     -     | Supported |     -     |     -     |     -     |
```

bridadan, 2016-08-08:
@zgoda Thanks for the explanation, that helps a lot!

@screamerbg This is most likely do to the use of `--mbedlib` as this pulls in the tools separately. This shouldn't be the case when using the mbed OS 5.1 release. Sounds like those tools just need to be updated?

screamerbg, 2016-08-29:
@zgoda The tools are now updated. Please recreate the program or alternatively delete the .temp folder in existing program and run `mbed deploy`. The upcoming version of mbed CLI (>=0.9.5) will automatically update the mbed Classic build tools.

screamerbg, 2016-09-06:
This is now fixed and mbed CLI automatically updates the mbed 2.0/Classic build tools when updating the codebase. Thanks for reporting this @zgoda

closed 2016-09-06 by screamerbg