ARMmbed/mbed-cli #322: Cannot handle empty mbed_app.json

LiyouZhou, 2016-08-08:
```
[mbed] Working path "/work/update-client-cliapp" (program)
[mbed] Exec "python -u /work/update-client-cliapp/mbed-os/tools/make.py -t GCC_ARM -m K64F --source . --build ./.build/K64F/GCC_ARM -v -j 1" in /work/update-client-cliapp
Error parsing './mbed_app.json':
Traceback (most recent call last):
  File "/work/update-client-cliapp/mbed-os/tools/make.py", line 271, in <module>
    name=options.artifact_name)
  File "/work/update-client-cliapp/mbed-os/tools/build_api.py", line 321, in build_project
    notify=notify, silent=silent, verbose=verbose, extra_verbose=extra_verbose, config=config)
  File "/work/update-client-cliapp/mbed-os/tools/build_api.py", line 241, in prepare_toolchain
    config = config or Config(target, src_paths)
  File "/work/update-client-cliapp/mbed-os/tools/config.py", line 211, in __init__
    self.app_config_data = json_file_to_dict(app_config_location) if app_config_location else {}
  File "/work/update-client-cliapp/mbed-os/tools/utils.py", line 254, in json_file_to_dict
    return dict_to_ascii(json.load(f, object_pairs_hook=OrderedDict))
  File "/usr/lib/python2.7/json/__init__.py", line 290, in load
    **kw)
  File "/usr/lib/python2.7/json/__init__.py", line 351, in loads
    return cls(encoding=encoding, **kw).decode(s)
  File "/usr/lib/python2.7/json/decoder.py", line 366, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/usr/lib/python2.7/json/decoder.py", line 384, in raw_decode
    raise ValueError("No JSON object could be decoded")
ValueError: No JSON object could be decoded
[mbed] ERROR: "python" returned error code 1.
[mbed] ERROR: Command "python -u /work/update-client-cliapp/mbed-os/tools/make.py -t GCC_ARM -m K64F --source . --build ./.build/K64F/GCC_ARM -v -j 1" in "/work/update-client-cliapp"
---
```

screamerbg, 2016-08-24:
CC @theotherjimmy

theotherjimmy, 2016-08-25:
So, this is actually an issue with `tools/config.py` in the mbed-os repo. However, an empty file is not a valid json object. I would expect that a file containing only `{}` to be valid.

bridadan, 2016-08-25:
@theotherjimmy Yeah that would be valid JSON. Though I think in this case, it'd probably be smart to go ahead and check if the JSON file is empty and issue a warning if it is.

screamerbg, 2016-08-29:
@theotherjimmy are you happy with the proposed solution by @bridadan ?

theotherjimmy, 2016-08-29:
Yes.

theotherjimmy, 2016-08-29:
please use https://github.com/ARMmbed/mbed-os/issues/2573 for discussion.

theotherjimmy, 2016-08-29:
@screamerbg can we close this issue now that It's tracked on mbed-os?

screamerbg, 2016-09-12:
@theotherjimmy Thanks for fixing this in mbed-os

closed 2016-09-12 by screamerbg