ARMmbed/mbed-cli #325: Python version requirements

tyomitch, 2016-08-16:
https://docs.mbed.com/docs/mbed-os-handbook/en/5.1/dev_tools/cli/ say that mbed CLI was tested with Python 2.7, without specifying the minor version.

I've observed that it doesn't work with Python versions older than 2.7.6; or with Python 3 altogether.

You may want to update the docs, or better yet, produce a diagnostic message when mbed CLI is run with an unsupported version of Python.

infinnovation, 2016-08-16:
The requirement for >= 2.7.9 is mentioned in https://docs.mbed.com/docs/mbedmicro-api/en/latest/api/md_README.html but I agree it should be in the document you cited.

In fact, I found that the attached patch allows me to run the mbed CLI with python 2.7.3

[regexp-fix.patch.txt](https://github.com/ARMmbed/mbed-cli/files/420489/regexp-fix.patch.txt)

screamerbg, 2016-08-18:
Thanks for reporting this

screamerbg, 2016-09-06:
This is now supported in the latest version of mbed CLI (0.9.5)

closed 2016-09-06 by screamerbg