ARMmbed/mbed-cli #326: Specifying -S option (--supported) should not require any other arguments

zgoda, 2016-08-17:
`-S` or `--supported` is informative only and should show the supported matrix regardless of other arguments for specific command, eg.:

```
$ mbed export -S
usage: mbed export [-h] -i IDE [-m TARGET] [--source SOURCE] [-c] [-S] [-v]
                   [-vv]
mbed export: error: argument -i/--ide is required
```

If I specify any IDE it fails other way:

```
$ mbed export -i GCC_ARM -S
[mbed] WARNING: Could not find mbed program in current path "/home/jazg/tmp".
[mbed] WARNING: You can fix this by calling "mbed new ." or "mbed config root ." in the root of your program.
---
[mbed] ERROR: The mbed tools were not found in "/home/jazg/tmp". 
[mbed] ERROR: Run `mbed deploy` to install dependencies and tools. 
---
```

I only wanted to see supported matrix. Ideally such informative operations should go to separate command that do not require any other arguments, say `mbed info` (like for example `docker info`).

screamerbg, 2016-08-24:
Thanks for reporting this.

Different versions of the OS support different platforms, e.g. the platforms support comes with mbed OS codebase, not with mbed CLI, therefore the mbed OS codebase is needed to correctly report what platforms are supported in that particular version of the OS.

screamerbg, 2016-09-06:
This is now supported in the latest version of mbed CLI (0.9.5). Again, thanks for reporting the issue.

closed 2016-09-06 by screamerbg