ARMmbed/mbed-cli #327: Building classic projects for microbit

arekzaluski, 2016-08-17:
Building classic projects for Microbit doesn't seem to work.
I've tried to build https://developer.mbed.org/teams/BBC/code/microbit_blinky/?platform=Microbit and https://developer.mbed.org/teams/microbit/code/microbit-hello-world/ projects. Both of them are failing to build correctly.

The command I'm using is 
`mbed compile -t GCC_ARM -m NRF51_MICROBIT`

@screamerbg @bogdanm

theotherjimmy, 2016-08-25:
Could you post the output that is associated with the failure after appending `-v` to the command line? It would help me root cause the problem you are having.

arekzaluski, 2016-08-25:
@theotherjimmy
Sure here is the output of
mbed compile -t GCC_ARM -m NRF51_MICROBIT -v

[microbit_output.txt](https://github.com/ARMmbed/mbed-cli/files/437971/microbit_output.txt)

janjongboom, 2016-09-05:
Potential micro:bit partner also ran into this... I cannot compile with ARMCC either.

Errors like this:

```
[Error] device_manager.h@509,37: 'p_ble_evt' was not declared in this scope
[ERROR] In file included from ./microbit/microbit-dal/nRF51822/nrf51-sdk/source/nordic_sdk/components/libraries/bootloader_dfu/dfu_app_handler.h:57:0,
                 from ./microbit/microbit-dal/BLE_API/ble/services/DFUService.h:26,
                 from ./microbit/microbit-dal/BLE_API/source/services/DFUService.cpp:19:
./microbit/microbit-dal/nRF51822/nrf51-sdk/source/nordic_sdk/components/ble/ble_services/ble_dfu/ble_dfu.h:190:44: error: 'ble_evt_t' has not been declared
 void ble_dfu_on_ble_evt(ble_dfu_t * p_dfu, ble_evt_t * p_ble_evt);
                                            ^
In file included from ./microbit/microbit-dal/nRF51822/nrf51-sdk/source/nordic_sdk/components/libraries/bootloader_dfu/dfu_app_handler.h:60:0,
                 from ./microbit/microbit-dal/BLE_API/ble/services/DFUService.h:26,
                 from ./microbit/microbit-dal/BLE_API/source/services/DFUService.cpp:19:
./microbit/microbit-dal/nRF51822/nrf51-sdk/source/nordic_sdk/components/ble/device_manager/device_manager.h:509:25: error: variable or field 'dm_ble_evt_handler' declared void
 void dm_ble_evt_handler(ble_evt_t * p_ble_evt);
                         ^
./microbit/microbit-dal/nRF51822/nrf51-sdk/source/nordic_sdk/components/ble/device_manager/device_manager.h:509:25: error: 'ble_evt_t' was not declared in this scope
./microbit/microbit-dal/nRF51822/nrf51-sdk/source/nordic_sdk/components/ble/device_manager/device_manager.h:509:37: error: 'p_ble_evt' was not declared in this scope
 void dm_ble_evt_handler(ble_evt_t * p_ble_evt);
```

bridadan, 2016-09-06:
I looked into why blinky wasn't compiling, looks like the pinned release of the microbit didn't actually support GCC_ARM at the time, or at least it wasn't included in the release. It compiles ok for ARM.

If you update the `mbed.bld` file to the latest version (`2e9cc70d1897`), I can compile blinky for GCC_ARM.

I'm afraid I don't know much about the other example since it isn't using the official mbed release.

arekzaluski, 2016-09-16:
@bridadan Thanks. I was able to build microbit-blinky after updating mbed.bld to the latest version.

screamerbg, 2016-10-18:
@arekzaluski Did this fix the issue? If so, are you happy to close this?

screamerbg, 2016-11-25:
Closing this as there is no activity. Please reopen if the issue persists.

0xc0170, 2018-08-01:
@jaustin Can you review this issue? The libraries are 2 years old by looks of it.

jaustin, 2018-08-01:
@theotherjimmy looked into this a while back, and I thought he had success with building the older programs. There was an inconsistency between the online compiler and the offline one WRT armcc, but that shouldn't be a factor here....  Jimmy, did you bump into any of the issues like Rob has?

theotherjimmy, 2018-08-01:
@jaustin, I did, and I resolved them with a build profile that was like `develop`, but without disabling VLA.

awe-coursera, 2018-08-02:
I also came across the kind of problem discussed above when trying to build a simple BBC:Microbit C++ program using the mbed-cli ... is there a simple set of instructions and example to show me how to do this please ...