ARMmbed/mbed-cli #328: [mbed] ERROR: Unknown Error: nothing to repeat

jsplyy, 2016-08-24:
[image]

hi:
Creating a new program for mbed OS 5 fail. how to deal with it?

screamerbg, 2016-08-24:
Thanks for reporting this. Do you think you could execute the same command with '-vv' and paste the output here? Thanks in advance.

smam-emb, 2016-08-26:
You need to use Python version >= 2.7.9. See the following link 
https://github.com/ARMmbed/mbed-cli/issues/325

screamerbg, 2016-09-14:
@jsplyy @smam-emb Is this still an issue with the latest mbed CLI (0.9.5)?

smam-emb, 2016-09-15:
For me this was resolved after downloading the corrrect Python version (including the minor version). Initially I had installed Python 2.7.

jsplyy, 2016-09-20:
Yes, when I update the python from 2.7.3 to python 2.7.12,it was resolved.

bridadan, 2016-10-11:
@jsplyy Glad you were able to find a solution! Closing.

closed 2016-10-11 by bridadan