ARMmbed/mbed-cli #329: Equivalent of yotta --config with mbed-cli

markus-becker-tridonic-com, 2016-08-24:
With yotta there was a possibility to select which config.json was used for the build (`yotta --config my-config.json`.

Is there a similar possibility with mbed-cli to choose a different file than mbed_app.json?

@MarceloSalazar @daniel-cesarini-tridonic-com

screamerbg, 2016-08-24:
CC @bogdanm

theotherjimmy, 2016-08-25:
We could support a similar option in the mbed-os tools, and propagate that option up to mbed-cli. It would be all on the config system though.

screamerbg, 2016-08-29:
@theotherjimmy do you think you could look at introducing the option in the mbed-os tools? Adding the help to mbed CLI is trivial really. Also, could you please create issue under https://github.com/ARMmbed/mbed-os?

screamerbg, 2016-09-12:
Implemented with https://github.com/ARMmbed/mbed-cli/pull/337. This would be introduced with the next mbed CLI release.

daniel-cesarini-tridonic-com, 2016-09-13:
@screamerbg @theotherjimmy Thank you. 

We will try this out as soon as it is ready and provide feedback on whether it solves our issue.

screamerbg, 2016-10-05:
I believe this is implemented and now live. Please reopen if that's not the case

closed 2016-10-05 by screamerbg

markus-becker-tridonic-com, 2016-10-24:
@screamerbg --app-config only works with an updated mbed-os, or?

bridadan, 2016-10-25:
@markus-becker-tridonic-com I believe this was added in version 5.1.4 of mbed OS. This was the PR that added this functionality: https://github.com/ARMmbed/mbed-os/pull/2593