ARMmbed/mbed-cli #333: Allowing --std=c++11 with GCC ARM tools.

nuket, 2016-08-29:
Is there any chance there could be an option to allow `--std=c++11` while compiling with the GCC_ARM tools?

screamerbg, 2016-08-30:
@sg- @theotherjimmy @geky @bogdanm

bridadan, 2016-08-30:
@theotherjimmy Your PR here would allow any arbitrary flags to be used, correct? https://github.com/ARMmbed/mbed-os/pull/2431

geky, 2016-08-30:
I personally would be very excited to adopt C++11. However, IAR does not support C++11 and is stuck at C++03 ([here](https://www.iar.com/iar-embedded-workbench/language-standards)). We simply can't adopt C++11 in mbed and related ecosystem without losing support for IAR.

_However_, applications are a different story. If an application writer intends for the application to only be used on specific toolchains, adopting C++11 in that application would be acceptable. Unfortunately, that would mean the application is not longer compatible with the rest of the mbed ecosystem or it would come with the false promise of support for IAR.

For this reason, we unfortunately can't encourage C++11. I do think we should allow it through @theotherjimmy's patch for the functionality, because in @theotherjimmy's patch it is clear you are working out of the mbed ecosystem.

nuket, 2016-08-30:
I'd be willing to assume the risk of it not working with IAR, at least at the application level.

geky, 2016-08-30:
Unfortunately there is currently no route to enabling C++11 short of modifying the tools ([this line](https://github.com/ARMmbed/mbed-os/blob/master/tools/toolchains/gcc.py#L41), if you were wondering).

@theotherjimmy has been working on exposing compiler flags in a less hacky manner:
https://github.com/ARMmbed/mbed-os/pull/2431

sg-, 2016-09-01:
mbed OS will not adopt C++11 as a default in the foreseeable future. When you export a makefile or into another IDE you're free to modify the build flags however you'd like (no compatibility guarantees). We're currently supporting C99 and C++03.

closed 2016-09-01 by sg-

bmcdonnell-ionx, 2018-01-18:
@geky, @sg-,

> I personally would be very excited to adopt C++11. However, IAR does not support C++11 and is stuck at C++03 ([here](https://www.iar.com/iar-embedded-workbench/language-standards)). We simply can't adopt C++11 in mbed and related ecosystem without losing support for IAR.

Looks like this is no longer true. From geky's linked source:

> ### ISO/ANSI C/C++ Compliance
> The IAR C/C++ compilers adhere to a freestanding implementation of the following C programming language standards:
>
> - ISO/IEC 14882:2015, known as C++14
> - ISO/IEC 9899:2012, known as C11
> - ANSI X3.159-1989, known as C89

Can this issue be revisited?

See also: https://www.iar.com/support/resources/articles/exploring-c11-and-c14/

> Support for these standards is available from version 8.10 of IAR Embedded Workbench for Arm.

Is mbed committed to supporting old versions of IAR? If so, how old?

Zooming out a bit... as an mbed OS user, I don't really care if mbed OS is built with C++11 enabled. (Can't think of a reason yet, anyway.) In some cases, I just want to be able to build my application with C++11. Thus I think if there were better support for using mbed OS as a library, this could be a non-issue.

sg-, 2018-01-22:
> Zooming out a bit... as an mbed OS user, I don't really care if mbed OS is built with C++11 enabled. (Can't think of a reason yet, anyway.) In some cases, I just want to be able to build my [...]

Why dont you make a local, custom, build profile and pass this to the tools? https://os.mbed.com/docs/v5.6/tools/build-profiles.html

bmcdonnell-ionx, 2018-01-22:
> Why dont you make a local, custom, build profile and pass this to the tools? 

Because I'm not building with `mbed compile`; I'm using `mbed export`, which AFAICT doesn't support build profiles.

theotherjimmy, 2018-01-22:
@bmcdonnell-ionx `mbed export` accepts `--profile`, but some exporters ignore it.

bmcdonnell-ionx, 2018-01-22:
I'm currently using NXP MCUXpresso, but will hopefully move to Visual Studio Code at some point.

IDE editor features are handy, and debugging is the big win. I'm not opposed to being *able* to `mbed compile`, but I don't want to be locked-in to that.

theotherjimmy, 2018-01-22:
@bmcdonnell-ionx That's not the way I would normally use "locked-in" :). I think MCUXpresso may ignore the `--profile` argument.