ARMmbed/mbed-cli #334: Why is the build folder hidden?

janjongboom, 2016-08-31:
I'm wondering why we hide the build folder? That's very user unfriendly. I've had one person in the forums who could not find his build folder (on Linux), and I just had to enable hidden folders on Windows just to be able to flash my board with a locally built program...

Suggestion: rename it to `build`, not `.build`; and make it non-hidden on Windows.

@screamerbg

screamerbg, 2016-08-31:
Historically the main reason is how directories are being handled in the build system, where `.build` is guaranteed to be excluded from target build (otherwise duplicate objects and so on). There will be a PR to both mbed OS and mbed CLI to fix this, but it would also require updates to:
* mbed OS 5 documentation and guides
* CI setups that use the default .build folder

This is already being discussed here https://github.com/ARMmbed/mbed-cli/issues/206

bridadan, 2016-10-11:
This was changed in version v0.9.7, can this be closed @janjongboom ?

janjongboom, 2016-10-11:
@bridadan Yep. Although a heads up would have been nice, as we'll need to update our docs...

screamerbg, 2016-10-19:
No longer hidden since mbed CLI 0.9.7

closed 2016-10-19 by screamerbg