ARMmbed/mbed-cli #338: Compile for debugging requires clean flag

arekzaluski, 2016-09-01:
Compile with debugging symbols requires -c (clean) flag to work while switching from standard build. mbed-cli should allow to build with debugging symbols without specifying additional clean flag. Currently it it is difficult to use it in IDE because it requires clean flag every time when it is being built for debugging.

bridadan, 2016-09-01:
This is similar to the issue raised here: https://github.com/ARMmbed/mbed-os/issues/2412

We'd basically need to store what compile options are used for each build and put it in the build directory, compare it with what the current build options are, and trigger a clean build if something changed.

screamerbg, 2016-09-05:
cc @theotherjimmy

screamerbg, 2016-10-19:
@arekzaluski This is now supported through the --profile option. Please reopen if this doesn't resolve your request.

closed 2016-10-19 by screamerbg