ARMmbed/mbed-cli #339: mbed-cli fails to compile due to long exec list

bremoran, 2016-09-01:
```
[DEBUG] Compile: arm-none-eabi-gcc -std=gnu99 -c -Wall -Wextra -Wno-unused-parameter -Wno-missing-field-initializers -fmessage-length=0 -fno-exceptions -fno-builtin -ffunction-sections -fdata-sections -funsigned-char -MMD -fno-delete-null-pointer-checks -fomit-frame-pointer -mcpu=cortex-m4 -mthumb -mfpu=fpv4-sp-d16 -mfloat-abi=softfp -g -O0 -DTARGET_KSDK2_MCUS -D__MBED__=1 -DDEVICE_I2CSLAVE=1 -D__FPU_PRESENT=1 -DTARGET_Freescale -DDEVICE_PORTINOUT=1 -DTARGET_RTOS_M4_M7 -DDEVICE_RTC=1 -DTOOLCHAIN_object -D__CMSIS_RTOS -DFSL_RTOS_MBED -DDEVICE_STORAGE=1 -DMBEDTLS_ENTROPY_HARDWARE_ALT -DTARGET_KPSDK_MCUS -DTOOLCHAIN_GCC -DMBED_BUILD_TIMESTAMP=1472721897.2 -DTARGET_CORTEX_M -DTARGET_DEBUG -DTARGET_LIKE_CORTEX_M4 -DDEVICE_ANALOGOUT=1 -DTARGET_M4 -DTARGET_UVISOR_UNSUPPORTED -DTARGET_K64F -DDEVICE_SERIAL=1 -DDEVICE_INTERRUPTIN=1 -DDEVICE_I2C=1 -DDEVICE_PORTOUT=1 -D__CORTEX_M4 -DDEVICE_STDIO_MESSAGES=1 -DCPU_MK64FN1M0VMD12 -DFEATURE_IPV4=1 -DTARGET_LIKE_MBED -DTARGET_FF_ARDUINO -DTARGET_KPSDK_CODE -DDEVICE_SERIAL_FC=1 -DFEATURE_STORAGE=1 -D__MBED_CMSIS_RTOS_CM -DDEVICE_SLEEP=1 -DTOOLCHAIN_GCC_ARM -DTARGET_FRDM -DDEVICE_SPI=1 -DDEVICE_ERROR_RED=1 -DDEVICE_SPISLAVE=1 -DDEVICE_ANALOGIN=1 -DDEVICE_PWMOUT=1 -DDEVICE_PORTIN=1 -DTARGET_MCU_K64F -DARM_MATH_CM4 @/Users/bremor01/dev/update/mbed-os-ucmm/.build/tests/K64F/GCC_ARM/.includes_baac15b9ac534c6d37c38ae45fb6ee89.txt -include /Users/bremor01/dev/update/mbed-os-ucmm/.build/tests/K64F/GCC_ARM/mbed_config.h -MD -MF /Users/bremor01/dev/update/mbed-os-ucmm/.build/tests/K64F/GCC_ARM/atomic-queue/source/atomic-cm3.d -o /Users/bremor01/dev/update/mbed-os-ucmm/.build/tests/K64F/GCC_ARM/atomic-queue/source/atomic-cm3.o /Users/bremor01/dev/update/mbed-os-ucmm/atomic-queue/source/atomic-cm3.c
        [DEBUG] Return: 1
        [DEBUG] Output: arm-none-eabi-gcc: error trying to exec '/usr/local/Cellar/arm-none-eabi-gcc/4.9-2015-q2-update/bin/../lib/gcc/arm-none-eabi/4.9.3/cc1': execv: Argument list too long
```

screamerbg, 2016-09-05:
cc @theotherjimmy

theotherjimmy, 2016-09-06:
Could you provide more details, like what Operating System you're testing this on, what version of gcc you're using, and the version of mbed-os? Those details would help me reproduce the problem on my end.

screamerbg, 2016-10-05:
Bump @bremoran @theotherjimmy

bremoran, 2016-10-12:
This was built on mac os. It was triggered by having a module with deeply nested directories (a yotta module) in the compilation path.

screamerbg, 2018-03-28:
I believe this is resolved. Please reopen if needed.

closed 2018-03-28 by screamerbg