ARMmbed/mbed-cli #340: Add NetBeans support to IDE project generator.

nuket, 2016-09-02:
NetBeans has really, really good support for dealing with Makefiles and code completion (including handling various preprocessor define settings) after parsing the `make` output. It also has really good support for setting up an ARM GCC cross-compiling toolchain.

Unfortunately, the Makefiles currently being generated by `mbed-cli` for mbed OS 5 projects aren't understood by NetBeans, it can't seem to find the source files. 

I only see errors like:
> $ make -f projectfiles/gcc_arm_NRF51_DK/Makefile 
make[1]: *** No rule to make target 'mbed-os/rtos/rtx/TARGET_CORTEX_M/TARGET_M0/TOOLCHAIN_GCC/SVC_Table.o', needed by 'Unnamed_Project.elf'.  Stop.
projectfiles/gcc_arm_NRF51_DK/Makefile:21: recipe for target 'all' failed
make: *** [all] Error 2

The old exporter from the mbed Classic online IDE could generate Makefiles that could be imported into NetBeans seamlessly. 

One difference seems to be that the Makefiles in Classic were generated into the root folder of the source code, where as the Makefiles in `mbed-cli` are generated under the `projectfiles` folder and somehow NetBeans can't pick up on how the `make` command is moving around the directory tree to get all of its sources.

I've attached the `Makefile`, `nbproject/project.xml`, and `nbproject/configurations.xml` for an mbed Classic app that I imported seamlessly into NetBeans, in case someone would be willing to take up the challenge of adding support for this for mbed OS 5 projects.

(That person could even be me, but I'd need some pointers on how the project generator gets called from mbed CLI.)

[project-files.zip](https://github.com/ARMmbed/mbed-cli/files/452026/project-files.zip)

screamerbg, 2016-09-05:
@sarahmarshy @0xc0170 @theotherjimmy

theotherjimmy, 2016-09-06:
FYI: I just tried an `mbed export -i gcc_arm -m K64F` within the `mbed-os-example-client` project with mbed-os checked out to https://github.com/ARMmbed/mbed-os/pull/2245 and imported the resulting project into netbeans. The compile worked.

nuket, 2016-09-07:
Ok, it works for the most part. I'm still testing.

Refactor requests: 

* If the export is for `gcc_arm`, it would be nice if the `GCC_BIN` setting in the `Makefile` were automatically set to the value of `GCC_ARM_PATH` in the `.mbed` file, if that value is already set for a project.
* Also, it would be nice if the `PROJECT` name in the `Makefile` could be settable by adding something like `mbed export -n <name>`.

There seemed to be a linker error to one of the functions I was calling from `mbedtls` in one of my projects, but I'll have to doublecheck that. It compiled file in Netbeans, but then did one of these numbers on me:

`Test.cpp:(.text._ZL7keyMeDov+0x46): undefined reference to 'mbedtls_pkcs5_pbkdf2_hmac'`

nuket, 2016-09-07:
Nevermind on that last point, I forgot to add -DMBED_PKCS5_C to the command line.

screamerbg, 2016-09-14:
cc @theotherjimmy

theotherjimmy, 2016-09-14:
>If the export is for gcc_arm, it would be nice if the GCC_BIN setting in the Makefile were automatically set to the value of GCC_ARM_PATH in the .mbed file, if that value is already set for a project.

This should be happening. See here: https://github.com/theotherjimmy/mbed/blob/1fb449cdd3ff7af47f684dd3a51586e011df6b7c/tools/export/makefile/__init__.py#L62-L65 The toolchain is being queried for it's compiler command. It looks like the problem lies with the toolchain GCC, as it does not set it's cc, cppc, etc. members to match the exact location of gcc in the FS.

If you have armc5 installed, you could export to make_armc5/make_iar (with the above PR) and see that it uses the toolchain paths (That would also be pretty cool: a NetBeans project compiling with armc5 or IAR).

>Also, it would be nice if the PROJECT name in the Makefile could be settable by adding something like mbed export -n <name>.

This seems reasonable, except that mbed export -n is already taken. I suggest `--name` instead. You're welcome to submit a counter-proposal for the flag, I'm not attached to `--name`.

screamerbg, 2016-10-19:
@nuket are you happy with the provided answer above? Also do you think this conversation can be taken over to https://github.com/ARMmbed/mbed-os instead?

nuket, 2016-10-19:
@screamerbg Yes, the export system has been working fine for me. I think this ticket could be closed.

theotherjimmy, 2016-12-07:
@screamerbg I just noticed that this issue is still open. Would you like to close it now?

closed 2017-11-28 by theotherjimmy