ARMmbed/mbed-cli #341: Error: C9912E: No --cpu selected

MarceloSalazar, 2016-09-07:
Priority: BLOCKER

Environment info:
Windows 7 (both cmd and git bash)
mbed CLI: 0.9.5
ARM Compiler 5.06 update 3 (build 300)
Keil / ARM Compiler installed here (default path): C:\Keil_v5\ARM\ARMCC
Windows system path points to C:\Keil_v5\ARM\ARMCC\bin

- Global config:
```
mbed config --list
[mbed] Global config:
No global configuration is set

[mbed] Local config (C:\Code\sandbox\memtests\mbed5\thread-testapp-private):
TOOLCHAIN=GCC_ARM
TARGET=K64F
```
Also tried to set mbed global ARM_PATH to `C:\Keil_v5\ARM\ARMCC` but didn't work.

- `mbed_settings.py`

Tried with and without `ARM_PATH = "C:/Keil_v5/ARM/ARMCC" `

- Error when invoking `mbed compile`
```
mbed compile -t ARM -m K64F -c
WARNING: MBED_ARM_PATH set as environment variable but doesn't exist
Building project thread-testapp-private (K64F, ARM)

then building as usual and then again same error:

Link: thread-testapp-private
Error: C9912E: No --cpu selected
Error: L6636E: Pre-processor step failed for './mbed-os/hal/targets/cmsis/TARGET_Freescale/TARGET_K64F/TOOLCHAIN_ARM_STD/MK64FN1M0xxx12.sct'
Error: L6372E: Image needs at least one load region.
Not enough information to list the image map.
Finished: 1 information, 0 warning and 2 error messages.
[ERROR] Error: C9912E: No --cpu selected
Error: L6636E: Pre-processor step failed for './mbed-os/hal/targets/cmsis/TARGET_Freescale/TARGET_K64F/TOOLCHAIN_ARM_STD/MK64FN1M0xxx12.sct'
Error: L6372E: Image needs at least one load region.
Not enough information to list the image map.
Finished: 1 information, 0 warning and 2 error messages.

[mbed] ERROR: "python" returned error code 1.
[mbed] ERROR: Command "python -u C:\Code\sandbox\memtests\mbed5\thread-testapp-private\mbed-os\tools\make.py -t ARM -m K64F --source . --build .\.build\K64F\ARM -c" in "C:\Code\sandbox\memtests\mbed5\thread-testapp-private"
```

@screamerbg @javier-moreno-tridonic-com

screamerbg, 2016-09-14:
Quick update:
@javier-moreno-tridonic-com @MarceloSalazar and myself tried to reproduce this unsuccessfully. We used identical:
* mbed OS code base
* ARM Compiler 5 (5.06 update3)
* mbed CLI
* Windows 7 and 10 platforms

Seems that the issue occurs only during the linking step.

screamerbg, 2016-10-05:
@javier-moreno-tridonic-com is this still an issue?

javier-moreno-tridonic-com, 2016-10-05:
@screamerbg I haven't tried it in a while. Today I'm booked but I could try to reproduce it again tomorrow

javier-moreno-tridonic-com, 2016-10-06:
@screamerbg Tried to compile again today with the master with the following setup:

thread-testapp-private

```
commit 9de036b6c347cdcb48e45e5745de75db46584046
Author: Arto Kinnunen <Arto.Kinnunen@arm.com>
Date:   Tue Sep 20 10:47:16 2016 +0300
```

Environment info:
Windows 7 (both cmd and git bash)
mbed CLI: 0.9.7 **updated**
ARM Compiler 5.06 update 3 (build 300)
Keil / ARM Compiler installed here (default path): C:\Keil_v5\ARM\ARMCC
Windows system path points to C:\Keil_v5\ARM\ARMCC\bin

the result was:

```
Link: thread-testapp-private
Error: C9912E: No --cpu selected
Error: L6636E: Pre-processor step failed for './mbed-os/hal/targets/cmsis/TARGET_Freescale/TARGET_K64F/TOOLCHAIN_ARM_STD/MK64FN1M0xxx12.sct'
Error: L6372E: Image needs at least one load region.
Not enough information to list the image map.
Finished: 1 information, 0 warning and 2 error messages.
[ERROR] Error: C9912E: No --cpu selected
Error: L6636E: Pre-processor step failed for './mbed-os/hal/targets/cmsis/TARGET_Freescale/TARGET_K64F/TOOLCHAIN_ARM_STD/MK64FN1M0xxx12.sct'
Error: L6372E: Image needs at least one load region.
Not enough information to list the image map.
Finished: 1 information, 0 warning and 2 error messages.

[mbed] ERROR: "python" returned error code 1.
[mbed] ERROR: Command "python -u C:\Code\sandbox\memtests\mbed5\thread-testapp-private\mbed-os\tools\make.py -t ARM -m K64F --source . --build .\BUILD\K64F\ARM -c" in "C:\Code\sandbox\memtests\mbed5\thread-testapp-private"
---
```

So it looks like the issue is still present

screamerbg, 2016-10-19:
@javier-moreno-tridonic-com Thanks for re-testing this. Don't think this is mbed CLI or mbed OS build tools issue. Do you think you could run `mbed compile ... -vv` and paste the linking command here? 

Also can you try the linking command locally? Just wandering whether the ARMCC5 linker will complain when executed directly.

javier-moreno-tridonic-com, 2016-10-20:
This is the linker output with the -vv option:
```
Link: thread-testapp-private
[DEBUG] Link: C:/Keil_v5/ARM/ARMCC\bin\armlink --via .\BUILD\K64F\ARM\.link_files.txt
[DEBUG] Return: 1
[DEBUG] Errors: Error: C9912E: No --cpu selected
[DEBUG] Errors: Error: L6636E: Pre-processor step failed for './mbed-os/hal/targets/cmsis/TARGET_Freescale/TARGET_K64F/TOOLCHAIN_ARM_STD/MK64FN1M0xxx12.sct'
[DEBUG] Errors: Error: L6372E: Image needs at least one load region.
[DEBUG] Errors: Not enough information to list the image map.
[DEBUG] Errors: Finished: 1 information, 0 warning and 2 error messages.
Traceback (most recent call last):
  File "C:\Code\sandbox\memtests\mbed5\thread-testapp-private\mbed-os\tools\make.py", line 271, in <module>
    name=options.artifact_name)
  File "C:\Code\sandbox\memtests\mbed5\thread-testapp-private\mbed-os\tools\build_api.py", line 353, in build_project
    res, _ = toolchain.link_program(resources, build_path, name)
  File "C:\Code\sandbox\memtests\mbed5\thread-testapp-private\mbed-os\tools\toolchains\__init__.py", line 939, in link_program
    self.link(elf, r.objects, r.libraries, r.lib_dirs, r.linker_script)
  File "C:\Code\sandbox\memtests\mbed5\thread-testapp-private\mbed-os\tools\hooks.py", line 31, in wrapper
    res = function(t_self, *args, **kwargs)
  File "C:\Code\sandbox\memtests\mbed5\thread-testapp-private\mbed-os\tools\toolchains\arm.py", line 209, in link
    self.default_cmd(cmd)
  File "C:\Code\sandbox\memtests\mbed5\thread-testapp-private\mbed-os\tools\toolchains\__init__.py", line 967, in default_cmd
    raise ToolException(_stderr)
ToolException: Error: C9912E: No --cpu selected
Error: L6636E: Pre-processor step failed for './mbed-os/hal/targets/cmsis/TARGET_Freescale/TARGET_K64F/TOOLCHAIN_ARM_STD/MK64FN1M0xxx12.sct'
Error: L6372E: Image needs at least one load region.
Not enough information to list the image map.
Finished: 1 information, 0 warning and 2 error messages.

[mbed] ERROR: "python" returned error code 1.
[mbed] ERROR: Command "python -u C:\Code\sandbox\memtests\mbed5\thread-testapp-private\mbed-os\tools\make.py -t ARM -m K64F --source . --build .\BUILD\K64F\ARM -c -v" in "C:\Code\sandbox\memtests\mbed5\thread-testapp-private"
---
```

MarceloSalazar, 2017-03-23:
@javier-moreno-tridonic-com do you continue to have this problem when using mbed CLI 1.0.0?

javier-moreno-tridonic-com, 2017-03-24:
@MarceloSalazar I haven't tried since October since there were no more feedback or pointers. Today and the following week I'll be very busy unfortunately, but after that I could try to reproduce the issue.

screamerbg, 2018-03-28:
I believe this is resolved. Please reopen if needed.

closed 2018-03-28 by screamerbg