ARMmbed/mbed-cli #342: mbed test - enhancement request - Could we limit the scope to a test group ?

adustm, 2016-09-12:
Hello, I've been reading https://github.com/ARMmbed/mbed-cli#limiting-the-test-scope

We would be happy to have the possibility to launch a tests family without adding the complete test names of each test_case.

Do you think it is feasible ?

Kind regards
Armelle
cc @svastm @bcostm

bridadan, 2016-09-12:
@adustm There is already the capability to launch a family of tests that have the same prefix by using a `*` character. You can run `mbed test -n tests-mbedmicro-rtos*` for instance to run all RTOS tests.

There has been ongoing work on Greentea to add support for "Test plans" that should provide more flexibility for executing a list of tests. The PR is open here: https://github.com/ARMmbed/greentea/pull/175

Would either of these features work for you?

adustm, 2016-09-13:
Hello Brian, 
Thank your for the trick, that's great and work fine.

I have created a PR to add this detail in the readme file, if you don't mind:
https://github.com/ARMmbed/mbed-cli/pull/345
You can use it directly or rephrase it the way you like, but could you please add this in the readme before I close this issue ?
Thanks in advance
Armelle

bridadan, 2016-09-13:
(Copying this here as well for clarity)

I made an alternative for this PR to clarify a few points, would mind taking a look here: #346

adustm, 2016-09-15:
LGTM

closed 2016-09-15 by adustm