ARMmbed/mbed-cli #343: mbed test : is it possible to use parallel compilation ?

adustm, 2016-09-12:
Hello,
We would like to use // compilation as we used to do it in mbed2. Is it possible ?

Kind regards
Armelle
cc @bcostm @svastm

bridadan, 2016-09-12:
It should be on by default (ex `-j 0` is the default setting for the test builder script). The only thing that isn't paralleled at the moment is linking the tests. Are you not seeing this to be the case? If you pass `-j 0` to `mbed test` do you see an improvement?

adustm, 2016-09-13:
```
mbed test -j 0 -m NUCLEO_F429ZI -t GCC_ARM
...
mbedgt-script.py: error: no such option: -j
[mbed] ERROR: "mbedgt" returned error code 2.
[mbed] ERROR: Command "mbedgt --test-spec C:\MCDGitW601\mbed\.build/tests\NUCLEO_F439ZI\GCC_ARM\test_spec.json -j 0" in "C:\MCDGitW601\mbed"
```

I have `mbedgt: greentea test automation tool ver. 1.2.0`
Is it too old ?

bridadan, 2016-09-13:
Ah, so the `mbed test` command will make a call to `mbed-os/tools/test.py` to do the compiling of the test. This accepts the `-j` parameter. However greentea (`mbedgt`) only runs the tests, so it does not accept a `-j` parameter. To limit the `mbed test` command to just compile, please add the `--compile` parameter.

adustm, 2016-09-15:
Thanks for your answer. We would like to know if it is possible to run the tests with parallel compilation as we used to do in mbed2.
If I understand correctly, I should rephrase with : could mbedgt support the -j option ?

bridadan, 2016-09-15:
@adustm Sorry I think I'm getting a bit mixed up :smile: 

If you want to **compile** the tests in parallel, meaning use more cpu cores to speed up just the build process of the tests, then this actually already occurs by default if you use the `mbed test` command. You shouldn't have to do anything different.

If you want to **run** the tests in parallel, meaning using multiple mbed boards plugged into your PC to speed up the actual execution of tests on hardware, there is currently experimental support for running tests this way in Greentea. But it has not been tested all that much and I wouldn't recommend it at the moment. If you're interested in that, I'd be happy to elaborate.

bridadan, 2016-10-11:
@adustm Can I help clear up anything else? If not, would you mind closing this?

bridadan, 2016-10-19:
I'm going to go ahead and close this, especially with the merge of https://github.com/ARMmbed/mbed-os/pull/2990.

Please reopen if there is still an issue!

closed 2016-10-19 by bridadan