ARMmbed/mbed-cli #344: mbed compile / mbed test: how to compile without clean option ?

adustm, 2016-09-12:
Hello,
mbed test is always rebuilding the complete mbed os database, as if we were using '-c' option.

Is it the normal behavior ? It takes long time while developping a new feature.
Could you check that or let us know how to avoid it, please ?

Kind regards
Armelle
cc @bcostm @svastm

screamerbg, 2016-09-14:
cc @bridadan

bridadan, 2016-09-14:
@adustm This is not normal behavior, I can confirm that this is a bug. It was introduced with this PR: https://github.com/ARMmbed/mbed-os/pull/2471/files

It makes it so it removes the `mbed_config.h` from the OS library build that is shared among all tests. This was to prevent there being multiple `mbed_config.h` from being in the build (one for the built test, and one for the shared OS library build). However, deleting this file triggers a clean build of the OS since the config state is being reset.

I need to examine this in more detail, but with the introduction of being able to specify app config on the command line we may be able to make smarter decisions about when to do a clean build.

screamerbg, 2016-10-05:
@adustm The PR @bridadan mentioned is now merged. Can you confirm that this is resolved?

screamerbg, 2016-10-19:
@adustm Closing as there is no activity on this. Please reopen if needed.

closed 2016-10-19 by screamerbg