ARMmbed/mbed-cli #347: Set macro NDEBUG for release builds

betzw, 2016-09-15:
It seems as if the macro NDEBUG gets never set even if doing a release build.

screamerbg, 2016-10-05:
@betzw Is this still relevant? I think this was fixed in the mbed OS build tools.

betzw, 2016-10-06:
After merging the latest version of the `master` branch I still do not see an automatic definition of macro `NDEBUG`.
Furthermore, the usage of flag `-o` seems now not working anymore :-1:

bridadan, 2016-10-06:
@betzw The -o flag has been replaced with the more general `--profiles` flag, you should be able to use that and it offers greater flexibility

betzw, 2016-10-07:
Maybe you should reflect this in the documentation including the help message which gets print be `mbed compile -h`.

bridadan, 2016-10-07:
Thanks @betzw, I submitted a PR here for just that: https://github.com/ARMmbed/mbed-cli/pull/365

I'm not as familiar with the `NDEBUG` option, so sorry I can't comment on it. Build profiles don't actually specify macros, just compiler flags. You should be able to set the `NDEBUG` macro through the config system.

pan-, 2016-10-07:
@bridadan 

>  Build profiles don't actually specify macros, just compiler flags.

So then it should be possible to add `-D NDEBUG` I guess.

bridadan, 2016-10-07:
Yup, `-D NDEBUG` should do the trick. You can also add it to your `mbed_app.json` config file if you don't feel like typing it on the command line. More info on application config is available here: https://github.com/ARMmbed/mbed-os/blob/master/docs/config_system.md#configuration-data-in-applications

pan-, 2016-10-07:
Sorry, I wasn't clear. I mean, `-D NDEBUG` **is** a compiler flag (for GCC at least, other compilers also support defines on the command line) so it should be possible to add it to a build profile instead of polluting the command line or `mbed_app.json`.

bridadan, 2016-10-07:
@pan- Oh you're absolutely right, my bad! Yeah you technically can do that since its just a compiler flag :smile: Though I would imagine we wouldn't be adding many (if any) compiler flags directly to the default build profiles since we then can't configure them. However with custom build profiles, you're free to do as you wish. Just know that things may break.

betzw, 2016-10-10:
As I can see you have defined a `debug.json` and a `default.json` build profile. Typically, macro `NDEBUG` should be defined for "release" builds (e.g. it removes the code for assertions), for which currently there is no _ecplicit_ profile in mbed-os.

bridadan, 2016-10-10:
The documentation for build profiles are here: https://github.com/ARMmbed/mbed-os/blob/master/docs/Toolchain_Profiles.md

I'm actually not 100% sure about the inheritance of the profiles. @theotherjimmy could you clarify this and maybe add it to the documentation?

screamerbg, 2016-10-19:
Ping @theotherjimmy

betzw, 2016-11-14:
Ping @theotherjimmy

closed 2017-01-02 by betzw