ARMmbed/mbed-cli #352: Problems with porting new boards

BlackstoneEngineering, 2016-09-20:
## Problem
When porting a new board it is possible to get into a situation where you have submitted mbed-os support and gotten mbed-os support mainlines, but not gotten mbed-ls support. It results in an in-ability to run tests.

### example
Delta board, currently it is listed in `mbed target --supported` as `DELTA_DFBM_NQ620`, but plugging one in and running detect or tests results in the following

#### `mbedls`
```
C:\cygwin\home\ausbla01\mbed\ci-test-shield>mbedls
+---------------+----------------------+-------------+-------------+--------------------------+-----------------+
| platform_name | platform_name_unique | mount_point | serial_port | target_id                | daplink_version |
+---------------+----------------------+-------------+-------------+--------------------------+-----------------+
| unknown       | None[0]              | D:          | unknown     | 45010310200164481E74B074 | 0310            |
+---------------+----------------------+-------------+-------------+--------------------------+-----------------+
```

#### `mbed detect`
```
C:\cygwin\home\ausbla01\mbed\ci-test-shield>mbed detect

[mbed] Detected None, port None, mounted D:
[mbed] Supported toolchains for None
Traceback (most recent call last):
  File "C:\cygwin\home\ausbla01\mbed\ci-test-shield\mbed-os\tools\detect_targets.py", line 77, in main
    print mcu_toolchain_matrix(platform_filter=r'^'+mut['mcu']+'$')
TypeError: cannot concatenate 'str' and 'NoneType' objects
[ERROR] cannot concatenate 'str' and 'NoneType' objects
[mbed] ERROR: "python" returned error code 1.
[mbed] ERROR: Command "python -u C:\cygwin\home\ausbla01\mbed\ci-test-shield\mbed-os\tools\detect_targets.py" in "C:\cygwin\home\ausbla01\mbed\ci-test
-shield"
```

#### `mbed test`
and trying to run `mbed test -n mbed-os-*` results in the following
```
Build successes:
  * DELTA_DFBM_NQ620::GCC_ARM::MBED-BUILD
  * DELTA_DFBM_NQ620::GCC_ARM::MBED-OS-FEATURES-FRAMEWORKS-UTEST-TESTS-UNIT_TESTS-BASIC_TEST
  * DELTA_DFBM_NQ620::GCC_ARM::MBED-OS-FEATURES-FRAMEWORKS-UTEST-TESTS-UNIT_TESTS-BASIC_TEST_DEFAULT
  * DELTA_DFBM_NQ620::GCC_ARM::MBED-OS-FEATURES-FRAMEWORKS-UTEST-TESTS-UNIT_TESTS-CASE_ASYNC_VALIDATE
  * DELTA_DFBM_NQ620::GCC_ARM::MBED-OS-FEATURES-FRAMEWORKS-UTEST-TESTS-UNIT_TESTS-CASE_CONTROL_ASYNC
  * DELTA_DFBM_NQ620::GCC_ARM::MBED-OS-FEATURES-FRAMEWORKS-UTEST-TESTS-UNIT_TESTS-CASE_CONTROL_REPEAT
  * DELTA_DFBM_NQ620::GCC_ARM::MBED-OS-FEATURES-FRAMEWORKS-UTEST-TESTS-UNIT_TESTS-CASE_SELECTION
  * DELTA_DFBM_NQ620::GCC_ARM::MBED-OS-FEATURES-FRAMEWORKS-UTEST-TESTS-UNIT_TESTS-CASE_SETUP_FAILURE
  * DELTA_DFBM_NQ620::GCC_ARM::MBED-OS-FEATURES-FRAMEWORKS-UTEST-TESTS-UNIT_TESTS-CASE_TEARDOWN_FAILURE
  * DELTA_DFBM_NQ620::GCC_ARM::MBED-OS-FEATURES-FRAMEWORKS-UTEST-TESTS-UNIT_TESTS-CONTROL_TYPE
  * DELTA_DFBM_NQ620::GCC_ARM::MBED-OS-FEATURES-FRAMEWORKS-UTEST-TESTS-UNIT_TESTS-MINIMAL_ASYNC_SCHEDULER
  * DELTA_DFBM_NQ620::GCC_ARM::MBED-OS-FEATURES-FRAMEWORKS-UTEST-TESTS-UNIT_TESTS-MINIMAL_SCHEDULER
  * DELTA_DFBM_NQ620::GCC_ARM::MBED-OS-FEATURES-FRAMEWORKS-UTEST-TESTS-UNIT_TESTS-TEST_ASSERTION_FAILURE_TEST_SETUP
  * DELTA_DFBM_NQ620::GCC_ARM::MBED-OS-FEATURES-FRAMEWORKS-UTEST-TESTS-UNIT_TESTS-TEST_SETUP_CASE_SELECTION_FAILURE
  * DELTA_DFBM_NQ620::GCC_ARM::MBED-OS-FEATURES-FRAMEWORKS-UTEST-TESTS-UNIT_TESTS-TEST_SETUP_FAILURE
  * DELTA_DFBM_NQ620::GCC_ARM::MBED-OS-RTOS-RTX-TARGET_CORTEX_M-TESTS-MEMORY-HEAP_AND_STACK
  * DELTA_DFBM_NQ620::GCC_ARM::MBED-OS-TESTS-INTEGRATION-BASIC
  * DELTA_DFBM_NQ620::GCC_ARM::MBED-OS-TESTS-MBEDMICRO-MBED-ATTRIBUTES
  * DELTA_DFBM_NQ620::GCC_ARM::MBED-OS-TESTS-MBEDMICRO-MBED-CALL_BEFORE_MAIN
  * DELTA_DFBM_NQ620::GCC_ARM::MBED-OS-TESTS-MBEDMICRO-MBED-CPP
  * DELTA_DFBM_NQ620::GCC_ARM::MBED-OS-TESTS-MBEDMICRO-MBED-DIV
  * DELTA_DFBM_NQ620::GCC_ARM::MBED-OS-TESTS-MBEDMICRO-RTOS-MBED-BASIC
  * DELTA_DFBM_NQ620::GCC_ARM::MBED-OS-TESTS-MBEDMICRO-RTOS-MBED-ISR
  * DELTA_DFBM_NQ620::GCC_ARM::MBED-OS-TESTS-MBEDMICRO-RTOS-MBED-MAIL
  * DELTA_DFBM_NQ620::GCC_ARM::MBED-OS-TESTS-MBEDMICRO-RTOS-MBED-MALLOC
  * DELTA_DFBM_NQ620::GCC_ARM::MBED-OS-TESTS-MBEDMICRO-RTOS-MBED-MUTEX
  * DELTA_DFBM_NQ620::GCC_ARM::MBED-OS-TESTS-MBEDMICRO-RTOS-MBED-QUEUE
  * DELTA_DFBM_NQ620::GCC_ARM::MBED-OS-TESTS-MBEDMICRO-RTOS-MBED-SEMAPHORE
  * DELTA_DFBM_NQ620::GCC_ARM::MBED-OS-TESTS-MBEDMICRO-RTOS-MBED-SIGNALS
  * DELTA_DFBM_NQ620::GCC_ARM::MBED-OS-TESTS-MBEDMICRO-RTOS-MBED-THREADS
  * DELTA_DFBM_NQ620::GCC_ARM::MBED-OS-TESTS-MBEDMICRO-RTOS-MBED-TIMER
  * DELTA_DFBM_NQ620::GCC_ARM::MBED-OS-TESTS-MBED_DRIVERS-C_STRINGS
  * DELTA_DFBM_NQ620::GCC_ARM::MBED-OS-TESTS-MBED_DRIVERS-DEV_NULL
  * DELTA_DFBM_NQ620::GCC_ARM::MBED-OS-TESTS-MBED_DRIVERS-ECHO
  * DELTA_DFBM_NQ620::GCC_ARM::MBED-OS-TESTS-MBED_DRIVERS-GENERIC_TESTS
  * DELTA_DFBM_NQ620::GCC_ARM::MBED-OS-TESTS-MBED_DRIVERS-LP_TIMEOUT
  * DELTA_DFBM_NQ620::GCC_ARM::MBED-OS-TESTS-MBED_DRIVERS-RACE_TEST
  * DELTA_DFBM_NQ620::GCC_ARM::MBED-OS-TESTS-MBED_DRIVERS-RTC
  * DELTA_DFBM_NQ620::GCC_ARM::MBED-OS-TESTS-MBED_DRIVERS-STL_FEATURES
  * DELTA_DFBM_NQ620::GCC_ARM::MBED-OS-TESTS-MBED_DRIVERS-TICKER
  * DELTA_DFBM_NQ620::GCC_ARM::MBED-OS-TESTS-MBED_DRIVERS-TIMEOUT
  * DELTA_DFBM_NQ620::GCC_ARM::MBED-OS-TESTS-MBED_DRIVERS-WAIT_US
  * DELTA_DFBM_NQ620::GCC_ARM::MBED-OS-TESTS-MBED_FUNCTIONAL-CALLBACK
  * DELTA_DFBM_NQ620::GCC_ARM::MBED-OS-TESTS-MBED_FUNCTIONAL-CALLBACK_BIG
  * DELTA_DFBM_NQ620::GCC_ARM::MBED-OS-TESTS-MBED_FUNCTIONAL-CALLBACK_SMALL
  * DELTA_DFBM_NQ620::GCC_ARM::MBED-OS-TESTS-MBED_FUNCTIONAL-FUNCTIONPOINTER
  * DELTA_DFBM_NQ620::GCC_ARM::MBED-OS-TESTS-MBED_HAL-LP_TICKER

Build skips:
  * DELTA_DFBM_NQ620::GCC_ARM::MBED-OS-TESTS-MBED_DRIVERS-MEM_TRACE
  * DELTA_DFBM_NQ620::GCC_ARM::MBED-OS-TESTS-MBED_DRIVERS-STATS
  * DELTA_DFBM_NQ620::GCC_ARM::MBED-OS-TESTS-STORAGE_ABSTRACTION-BASICAPI
mbedgt: greentea test automation tool ver. 1.2.0
mbedgt: test specification file 'C:\cygwin\home\ausbla01\mbed\ci-test-shield\.build/tests\DELTA_DFBM_NQ620\GCC_ARM\test_spec.json' (specified with --t
est-spec option)
mbedgt: using 'C:\cygwin\home\ausbla01\mbed\ci-test-shield\.build/tests\DELTA_DFBM_NQ620\GCC_ARM\test_spec.json' from current directory!
mbedgt: detecting connected mbed-enabled devices...
mbedgt: detected 1 device
mbedgt: mbed-ls was unable to enumerate correctly all properties of the device!
        check with 'mbedls -j' command if all properties of your device are enumerated properly
mbedgt: mbed-ls property 'serial_port' is 'None'
        check if your serial port driver is correctly installed!
mbedgt: mbed-ls property 'platform_name' is 'None'
mbedgt: processing target 'DELTA_DFBM_NQ620' toolchain 'GCC_ARM' compatible platforms... (note: switch set to --parallel 1)
mbedgt: all tests finished!
mbedgt: shuffle seed: 0.6793174890
mbedgt: no platform/target matching tests were found!
mbedgt: no matching platforms were found!
mbedgt: completed in 0.67 sec
mbedgt: exited with code -110
[mbed] ERROR: "mbedgt" returned error code -110.
[mbed] ERROR: Command "mbedgt --test-spec C:\cygwin\home\ausbla01\mbed\ci-test-shield\.build/tests\DELTA_DFBM_NQ620\GCC_ARM\test_spec.json -n mbed-os-
*" in "C:\cygwin\home\ausbla01\mbed\ci-test-shield"
---
```

## Solution
1. fix the `detect` command to handle values of `None`, cause it looks like that is what its trying to lookup, which causes issues. I would suggest prompting the user with the command to manually patch / alias the board. 
2. ??

BlackstoneEngineering, 2016-09-20:
@theotherjimmy

theotherjimmy, 2016-09-20:
>fix the detect command to handle values of None, cause it looks like that is what its trying to lookup, which causes issues. I would suggest prompting the user with the command to manually patch / alias the board. 

We can't always do user input, as it would hang something like a CI job.

BlackstoneEngineering, 2016-09-20:
Ah, no, not requiring user input, more like an error message that suggests a fix the user can try next time. Thinking as a post failure debug step, not a command interrupt step.

screamerbg, 2016-09-21:
@theotherjimmy Think the python error is first priority (detect_targets.py). Second is handling of unknown targets in the test tools.

cc @bridadan

bridadan, 2017-01-12:
An error message with something like "Target not found" and then listing possible solutions like:

- Is the board plugged in?
- Are you developing a new board? Try mocking the device: https://github.com/ARMmbed/mbed-ls#mocking-new-or-existing-target-to-custom-platform-name (or just list the command `mbedls --mock <first 4 characters of TargetID>:DELTA_DFBM_NQ620`)

screamerbg, 2017-12-02:
Bump @bridadan @theotherjimmy What's the latest status on this?

bridadan, 2017-12-04:
Hi @screamerbg. Phew, this is an old issue! Looks like the tools really just need a more helpful error message? Maybe adding those two points I mentioned above?

I'm still getting caught up but I may get a PR up with those changes so we can close this out.

screamerbg, 2018-03-28:
Mbed CLI now provides the following message if USB devices are detected, but the target is not recognized.

```
If you're developing a new target, you can mock the device to continue your development. 
Use 'mbedls --mock ID:NAME' to do so (see 'mbedls --help' for more information)
```

@BlackstoneEngineering @bridadan does this solve the feature request? (bug is already fixed)

screamerbg, 2018-04-28:
Closing this issue as the bug was fixed. Let's follow-up in #578 regarding Cygwin support. In my environment Cygwin v2.10 (latest) for Windows 10 64bit, Mbed CLI works just fine.

closed 2018-04-28 by screamerbg