ARMmbed/mbed-cli #353: Upgrade from 0.9.0 to 0.9.5

sg-, 2016-09-22:
OS Error but the upgrade seemed to happen OK.
```
(yotta workspace) ~ $ mbed --version
0.9.0
(yotta workspace) ~ $ sudo pip install -U mbed-cli
Password:
The directory '/Users/samgro01/Library/Caches/pip/http' or its parent directory is not owned by the current user and the cache has been disabled. Please check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
The directory '/Users/samgro01/Library/Caches/pip' or its parent directory is not owned by the current user and caching wheels has been disabled. check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
Collecting mbed-cli
  Downloading mbed-cli-0.9.5.zip (48kB)
    100% |████████████████████████████████| 49kB 1.2MB/s 
Installing collected packages: mbed-cli
  Found existing installation: mbed-cli 0.9.0
    Uninstalling mbed-cli-0.9.0:
      Successfully uninstalled mbed-cli-0.9.0
  Running setup.py install for mbed-cli
Successfully installed mbed-cli
Traceback (most recent call last):
  File "/Applications/yotta.app/Contents/Resources/workspace/bin/pip", line 11, in <module>
    sys.exit(main())
  File "/Applications/yotta.app/Contents/Resources/workspace/lib/python2.7/site-packages/pip/__init__.py", line 217, in main
    return command.main(cmd_args)
  File "/Applications/yotta.app/Contents/Resources/workspace/lib/python2.7/site-packages/pip/basecommand.py", line 248, in main
    pip_version_check(session)
  File "/Applications/yotta.app/Contents/Resources/workspace/lib/python2.7/site-packages/pip/utils/outdated.py", line 102, in pip_version_check
    installed_version = get_installed_version("pip")
  File "/Applications/yotta.app/Contents/Resources/workspace/lib/python2.7/site-packages/pip/utils/__init__.py", line 858, in get_installed_version
    working_set = pkg_resources.WorkingSet()
  File "/Applications/yotta.app/Contents/Resources/workspace/lib/python2.7/site-packages/pip/_vendor/pkg_resources/__init__.py", line 629, in __init__
    self.add_entry(entry)
  File "/Applications/yotta.app/Contents/Resources/workspace/lib/python2.7/site-packages/pip/_vendor/pkg_resources/__init__.py", line 685, in add_entry
    for dist in find_distributions(entry, True):
  File "/Applications/yotta.app/Contents/Resources/workspace/lib/python2.7/site-packages/pip/_vendor/pkg_resources/__init__.py", line 2075, in find_eggs_in_zip
    if metadata.has_metadata('PKG-INFO'):
  File "/Applications/yotta.app/Contents/Resources/workspace/lib/python2.7/site-packages/pip/_vendor/pkg_resources/__init__.py", line 1605, in has_metadata
    return self.egg_info and self._has(self._fn(self.egg_info, name))
  File "/Applications/yotta.app/Contents/Resources/workspace/lib/python2.7/site-packages/pip/_vendor/pkg_resources/__init__.py", line 1963, in _has
    return zip_path in self.zipinfo or zip_path in self._index()
  File "/Applications/yotta.app/Contents/Resources/workspace/lib/python2.7/site-packages/pip/_vendor/pkg_resources/__init__.py", line 1843, in zipinfo
    return self._zip_manifests.load(self.loader.archive)
  File "/Applications/yotta.app/Contents/Resources/workspace/lib/python2.7/site-packages/pip/_vendor/pkg_resources/__init__.py", line 1783, in load
    mtime = os.stat(path).st_mtime
OSError: [Errno 2] No such file or directory: '/Applications/yotta.app/Contents/Resources/workspace/lib/python2.7/site-packages/mbed_cli-0.9.0-py2.7.egg'
(yotta workspace) ~ $ mbed
usage: mbed [-h] [--version]             ...

Command-line code management tool for ARM mbed OS - http://www.mbed.com
version 0.9.5
```

screamerbg, 2016-09-23:
@sg- Often pip generates error when switching between public pip modules (e.g. `pip install ...`) and manually installed modules (e.g. `setup.py install`) . Can you please retry and let me know?

screamerbg, 2016-10-05:
Ping @sg-

sg-, 2016-10-05:
Given that 0.9.0 doesn't exist in pip it must have been installed by setup.py locally.

```
Collecting mbed-cli==0.9.0
  Could not find a version that satisfies the requirement mbed-cli==0.9.0 (from versions: 0.5.1, 0.5.2, 0.6.8, 0.7.1, 0.7.2, 0.7.3, 0.7.13,
0.7.17, 0.8.0, 0.8.1, 0.8.2, 0.8.5, 0.8.7, 0.8.8, 0.8.9, 0.9.1, 0.9.5, 0.9.7)
No matching distribution found for mbed-cli==0.9.0
```

screamerbg, 2016-10-06:
@sg- Not sure how we can solve the compatibility between locally installed packages and public ones. Seems that mixing these causes pip to throw some nasty errors.

sg-, 2016-10-06:
Seems reasonable.

closed 2016-10-06 by sg-