ARMmbed/mbed-cli #358: require compiler and library support for the ISO C++ 2011 standard.

fwndz, 2016-09-29:
Hi, I need some help, I am using mbed cli for compiling my program. When I run simple blink led code it worked fine. But when i try to run mqtt example from here (https://developer.mbed.org/teams/ESP8266/code/ESP8266_MQTT_HelloWorld/) error occured. I run the compiler in ubuntu

`[Error] c++0x_warning.h@32,2: #error This file requires compiler and library support for the ISO C++ 2011 standard. This support is currently experimental, and must be enabled with the -std=c++11 or -std=gnu++11 compiler options.
[Error] Endpoint.h@69,15: declaration of 'ESP8266* Endpoint::ESP8266' [-fpermissive]
`

As referring to error message above, it shows that "This file requires compiler and library support for the ISO C++ 2011 standard", Can some one help me to solve the problem? How should I do to enable C++2011 std.?

Thanks

bridadan, 2016-09-29:
Hi @fwndz, I was able to reproduce this issue locally as well. I have worked with that example in the past, and I believe it was really only tested when compiling with the online compiler. So the GCC support may be spotty.

If you have the ARM toolchain installed (or the uVision IDE) then you should be able to use that to compile the example. I tried doing that with mbed CLI and it compiled OK. I was also able to compile it in the online compiler.

BTW, that's a fairly old example, so you may find it a bit out of date. There's ESP8266 support in mbed OS 5, though I think MQTT hasn't had much attention. Sure would be cool to see that working though :smile: 

If you want to follow up more on this, you may have more luck on the [developer site](http://developer.mbed.org), as this issue is more related to the example code vs the actual tool mbed CLI.

fwndz, 2016-09-29:
Hi @bridadan , Thanks for the reply, BTW I was able to compile it on mbed OS 5 and uVision. However, when I tried to use mbed cli that error occur. I do hope I can compile the example with mbed CLI. Do you have any idea how to solve the problem?

bridadan, 2016-09-29:
@fwndz Are you using the ARM or GCC_ARM toolchain when compiling with mbed CLI? Can you share the command you're using? And perhaps share the output of : `mbed config -L`

fwndz, 2016-09-29:
@bridadan  I use GCC_ARM toolchain

```
[mbed]Global config:
No global configuration is set
[mbed] Local config (/home/fawwaz/mbedProject/ESP8266_MQTT_HelloWorld):
TOOLCHAIN=GCC_ARM
TARGET=NUCLEO_F411RE
```
`

bridadan, 2016-09-29:
Ah yeah, so that's the issue. I also found that the example doesn't compile with GCC_ARM.

If you have uVision installed already, try running the following command:

```
mbed compile -t ARM
```

If that doesn't work, you'll need to run the following command first before retrying (you may have to update the path depending on where you installed uVision):

```
mbed config --global ARM_PATH "C:/Keil_v5/ARM/ARMCC"
```

Then try the `mbed compile -t ARM` command again.

I know uVision comes with different versions of the ARM Compiler, so you may have mixed results with this, but it's definitely worth a try!

fwndz, 2016-09-29:
Thanks @bridadan. If I want to compile  it in on ubuntu? How can I get ARM toolchain?

bridadan, 2016-09-29:
Ah, that becomes more difficult. This is where my knowledge tends to get a bit thin :)

You can download the standalone compiler here (though I think this requires the commercial license): https://developer.arm.com/products/software-development-tools/compilers/arm-compiler-5/downloads

Not sure if it's code size limited or if it just straight up won't work, I've only used it with a license.

Another workaround for your problem may be to alter the code a bit. Pretty sure the error you posted above stems from this file including `cstdint`: https://developer.mbed.org/teams/ESP8266/code/ESP8266Interface/file/03fd9333670d/Socket/UDPSocket.h

Maybe you could rework it to remove that dependency?

fwndz, 2016-09-29:
@bridadan thanks for the great support :+1:

bridadan, 2016-09-29:
@fwndz Happy to help! Would you mind closing this if you have a path forward?

closed 2016-09-29 by fwndz