ARMmbed/mbed-cli #360: "mbed update" does not update mbed-os

infinnovation, 2016-10-05:
It appears that "mbed update" does not actually update the mbed-os library - it changes the hash in the mbed-os.lib file but does not checkout the corresponding version.

```ShellSession
jac-16:~/mbed/bls-test $ mbed --version
0.9.7
jac-16:~/mbed/bls-test $ mbed ls
bls-test (bda57f050ffd)
|- bls (a0caf94e9d38)
`- mbed-os (21dd7008a154)
jac-16:~/mbed/bls-test $ mbed update --clean --clean-deps
[mbed] Synchronizing dependency references...
[mbed] Updating program "bls-test" to latest revision in the current branch
[mbed] Updating library "mbed-os" to rev #a6f3fd1a60d5
[mbed] Updating library "bls" to rev #a0caf94e9d38
jac-16:~/mbed/bls-test $ 
jac-16:~/mbed/bls-test $ mbed ls
bls-test (bda57f050ffd)
|- bls (a0caf94e9d38)
`- mbed-os (21dd7008a154)
```

screamerbg, 2016-10-05:
@infinnovation can you provide me with an example that I can reproduce locally? Perhaps the bls-test program (without the source code)?

infinnovation, 2016-10-06:
I've set both bls-test and bls as Public (Unlisted) to allow you to fetch them.  And in testing that, I have a simpler anomaly:

```ShellSession
jac-16:~/mbed $ mbed import https://developer.mbed.org/users/infinnovation/code/bls-test
[mbed] Importing program "bls-test" from "https://developer.mbed.org/users/infinnovation/code/bls-test" at latest revision in the current branch
[mbed] Adding library "mbed-os" from "https://github.com/ARMmbed/mbed-os" at rev #a6f3fd1a60d5
[mbed] Adding library "bls" from "https://developer.mbed.org/users/infinnovation/code/bls" at rev #a0caf94e9d38
jac-16:~/mbed $ 
jac-16:~/mbed $ cd bls-test
jac-16:~/mbed/bls-test $ mbed ls
bls-test (bda57f050ffd)
|- bls (a0caf94e9d38)
`- mbed-os (21dd7008a154)
```

Looks like it's something to do with the cache.  I removed cache/github.com/ARMmbed/mbed-os and tried again; this time it got the proper version of mbed-os

screamerbg, 2016-10-06:
@infinnovation Have you enabled the experimental caching in mbed CLI?

infinnovation, 2016-10-06:
Yes, I have enabled caching, and I was aware that it is experimental.  Perhaps the two test-cases I've inadvertently generated will help its advancement towards fully working :-)

The .git repo in the cache (before I removed it) seemed to have been fully "fetched" (including tag mbed-os-5.1.5) but HEAD was at 21dd700...

screamerbg, 2016-10-18:
@infinnovation PR #374 is aiming to fix this. Think you could try it?

infinnovation, 2016-10-18:
@screamerbg I've tried it out and it works fine.

screamerbg, 2016-10-20:
Now merged and part of the mbed CLI 0.9.10 release. Please reopen if the issue persists.

closed 2016-10-20 by screamerbg