ARMmbed/mbed-cli #362: "mbed publish" sends to a 404 in case the project is not Git-enabled

kartben, 2016-10-06:
Steps to reproduce:

```
mbed new test
cd test
mbed publish
```

This outputs the following message:

```
[mbed] Checking for local modifications...
[mbed] ERROR: Program "test" in "/private/tmp/test" is a local repository.
[mbed] ERROR: Please associate it with a remote repository URL before attempting to publish.
[mbed] ERROR: Read more about git repositories here:
[mbed] ERROR: http://developer.mbed.org/handbook/how-to-publish-with-git/
```

http://developer.mbed.org/handbook/how-to-publish-with-git/ is a 404.

screamerbg, 2016-10-19:
Fixed in https://github.com/ARMmbed/mbed-cli/pull/379

screamerbg, 2016-10-20:
Now merged and part of the mbed CLI 0.9.10 release. Please reopen if the issue persists.

closed 2016-10-20 by screamerbg