ARMmbed/mbed-cli #363: Stm32F CPU STOP issue

Sim-jaeil, 2016-10-06:
```c++
#include "mbed.h"
Serial pc(PA_2, PA_3);

void Callback(){
 char Temp_Data = pc.getc();
 pc.putc(Temp_Data);
}

int main(){
 pc.attach(&Callback,Serial::RxIrq);
 while(1){
  
 }
}
```
I am currently using the device Nucleo_F401RE.
But even the same symptoms MBED 2.0, the CPU will stop the moment it takes an interrupt is received.
The compiler is okay to stop using the Microlib the Keil compiler is online and, Keil. It seems there is a problem with the MBED.

Until now Export by an external compiler, but resolved using Microlib The MicroLib there are additional libraries that do not work are looking for what to do now resolved.

Please help.

screamerbg, 2016-10-06:
@Sim-jaeil are you happy to file the issue against the https://github.com/ARMmbed/mbed-os repository where all the boards/platforms code live?

screamerbg, 2016-10-19:
Ping @Sim-jaeil

closed 2016-10-19 by Sim-jaeil