ARMmbed/mbed-cli #364: Not installing correclty on Ubuntu 16

BlackstoneEngineering, 2016-10-06:
## Problem
On Ubuntu 16 you can install mbed-cli from pip, but then mbed-cli is not recognized by the system as a valid command. Instead I must drill down into the `~/.local/lib/python2.7/site-packages/mbed` folder and call the file specifically. 

Output looks like the following
```shell
 ~$ mbed -h
mbed: command not found

 ~$ pip --version
pip 8.1.2 from /home/kim/.local/lib/python2.7/site-packages (python 2.7)  

~$ python --version 
Python 2.7.12  

 ~$ pip install mbed-cli
Requirement already satisfied (use --upgrade to upgrade): mbed-cli in ./.local/lib/python2.7/site-packages

 ~$ ls ./.local/lib/python2.7/site-packages/mbed
__init__.py  __init__.pyc  mbed.py  mbed.pyc  

~$ python ./.local/lib/python2.7/site-packages/mbed/mbed.py -h
usage: mbed [-h] [--version] 

Command-line code management tool for ARM mbed OS - http://www.mbed.com version 0.9.7

Use 'mbed <command> -h|--help' for detailed help.
Online manual and guide available at https://github.com/ARMmbed/mbed-cli

optional arguments:
  -h, --help   show this help message and exit
  --version    print version number and exit

Commands:
             
    new        Create new mbed program or library
    import     Import program from URL
    add        Add library from URL
    remove     Remove library
    deploy     Find and add missing libraries
    publish    Publish program or library
    update     Update to branch, tag, revision or latest
    sync       Synchronize library references
    ls         View dependency tree
    status     Show version control status
               
    compile    Compile code using the mbed build tools
    test       Find, build and run tests
    export     Generate an IDE project
    detect     Detect connected mbed targets/boards
               
    config     Tool configuration
    target     Set or get default target
    toolchain  Set or get default toolchain
               
    help       This help screen
 ```

## Solution

It is unknown if this is a change in how Ubuntu16 is handling pythong packages or if its something else. Either way it seems that vanilla Ubuntu16 will not work and we either need to account for a paradigm shift in how ubuntu is treating pip, or add a workaround to our tool

screamerbg, 2016-10-19:
@BlackstoneEngineering Interestingly, I got similar report from @ansondtx20 regarding Mac OSX with the very latest Homebrew. Seems that mbed CLI is installed correctly but the Python/bin folder is not in the system path. Can you check whether that's the case on the Ubuntu16?

BlackstoneEngineering, 2016-10-19:
It was not in the system path, which was really weird.

screamerbg, 2016-10-19:
@BlackstoneEngineering There is a "hack" for linux systems where we can introduce the shell file during the package setup process. Not sure this is great workaround though. Also not having Python/pip bin path to your system path means that you shouldn't be able to run mbedls as well, right?

BlackstoneEngineering, 2016-10-19:
Need to dig further in here, will try to spin up a VM in the next couple days to re-duplicate and reconfirm. 

I realize I didnt mention it, but this error was reported by an internal employee, so I raised it here to not forget. Will duplicate and confirm shortly.

gtsiros, 2016-11-10:
Tried following the video tutorial (which wastes a lot of time for nothing)

Of course it doesn't work (with the level of segmentation and lack of homogenous structure, it can only be expected)

python being installed in weird directories or something (what's a .local ?)

there's a #define <new> somewhere in mbed.h, i think
removing that, there is a slew of errors that i can't even make out because they are formatted like line noise in the terminal. arm-gcc throws an error, python reports on it, after that everything goes to hell

I spent more time trying to wade through crap (git project initializations, python plugins installations and whatnot) than actual compilation.

I don't get why it is considered preferable to wrap up the already wrapped up build process (make) with more scripts (mbed-cli compile), use two versioning systems and two scripting languages (python and bash) 

I don't know if mbed is a community effort or a product, but it is a sad let-down. In the time and effort it takes to even get *started* with it, I can _download, install, run and compile_ programs with (say) lazarus.

kimphill, 2016-11-16:
Hi, I'm the other employee BlackstoneEngineering mentioned.  I tried this on another Ubuntu 16.04.1 machine, and it still fails.  Might I add I thought Ubuntu had started migrating from installing things python away from pip and to apt?  'apt search mbed' returns nothing, however.  There is a re-check label below on someone asking whether mbedls is also not in the path, and that is correct (I get "mbedls: command not found"). Clearly mbed-cli has integration issues with at least Ubuntu 16.04.1:  is it not a supported Ubuntu version?  Details - similar to those BlackstoneEngineering originally reported but this time on a different 16.04.1 machine to prove it's not the original machine's fault - follow:

------------------------------------------------------------------------
$ mbed -h
No command 'mbed' found, did you mean:
 Command 'med' from package 'ncl-ncarg' (universe)
 Command 'mred' from package 'racket' (universe)
mbed: command not found
$ pip --version
pip 8.1.1 from /usr/lib/python2.7/dist-packages (python 2.7)
$ python --version
Python 2.7.12
$ pip install mbed-cli
Collecting mbed-cli
  Downloading mbed-cli-0.9.10.zip (48kB)
    100% |████████████████████████████████| 51kB 2.6MB/s 
Building wheels for collected packages: mbed-cli
  Running setup.py bdist_wheel for mbed-cli ... done
  Stored in directory: /home/kimphi01/.cache/pip/wheels/4d/ed/d5/e989b2bdad22c879f77f5c8f1e2ce7d82300790dddc732025d
Successfully built mbed-cli
Installing collected packages: mbed-cli
Successfully installed mbed-cli-0.9.10
You are using pip version 8.1.1, however version 9.0.1 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
$ mbed -h
No command 'mbed' found, did you mean:
 Command 'mred' from package 'racket' (universe)
 Command 'med' from package 'ncl-ncarg' (universe)
mbed: command not found
$ python ./.local/lib/python2.7/site-packages/mbed/mbed.py -h
usage: mbed [-h] [--version]             ...

Command-line code management tool for ARM mbed OS - http://www.mbed.com
version 0.9.10

Use 'mbed <command> -h|--help' for detailed help.
Online manual and guide available at https://github.com/ARMmbed/mbed-cli

optional arguments:
  -h, --help   show this help message and exit
  --version    print version number and exit

Commands:
             
    new        Create new mbed program or library
    import     Import program from URL
    add        Add library from URL
    remove     Remove library
    deploy     Find and add missing libraries
    publish    Publish program or library
    update     Update to branch, tag, revision or latest
    sync       Synchronize library references
    ls         View dependency tree
    status     Show version control status
               
    compile    Compile code using the mbed build tools
    test       Find, build and run tests
    export     Generate an IDE project
    detect     Detect connected mbed targets/boards
               
    config     Tool configuration
    target     Set or get default target
    toolchain  Set or get default toolchain
               
    help       This help screen
$ apt search mbed | grep -w mbed

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

$ cat /etc/issue
Ubuntu 16.04.1 LTS \n \l

$ 
------------------------------------------------------------------------

MarceloSalazar, 2016-11-25:
I also hit this issue when using pip 9.0.1 (latest).
After downgrading pip to 8.1.1, now mbed CLI 1.0.0 works like a charm

ashok-rao, 2016-11-25:
Issue with pip, not Ubuntu. Running the following on a VM..
```
root@ubuntu:/usr/Ashok/mbed_CLI/blinky/oob-mbed-os-example-blinky# cat /etc/os-release
NAME="Ubuntu"
VERSION="16.04.1 LTS (Xenial Xerus)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 16.04.1 LTS"
VERSION_ID="16.04"
HOME_URL="http://www.ubuntu.com/"
SUPPORT_URL="http://help.ubuntu.com/"
BUG_REPORT_URL="http://bugs.launchpad.net/ubuntu/"
UBUNTU_CODENAME=xenial
root@ubuntu:/usr/Ashok/mbed_CLI/blinky/oob-mbed-os-example-blinky# python --version
Python 2.7.12
root@ubuntu:/usr/Ashok/mbed_CLI/blinky/oob-mbed-os-example-blinky# pip --version
pip 8.1.1 from /usr/lib/python2.7/dist-packages (python 2.7)
root@ubuntu:/usr/Ashok/mbed_CLI/blinky/oob-mbed-os-example-blinky# mbed
usage: mbed [-h] [--version]             ...

Command-line code management tool for ARM mbed OS - http://www.mbed.com
version 1.0.0

Use 'mbed <command> -h|--help' for detailed help.
Online manual and guide available at https://github.com/ARMmbed/mbed-cli

optional arguments:
  -h, --help   show this help message and exit
  --version    print version number and exit

Commands:
             
    new        Create new mbed program or library
    import     Import program from URL
    add        Add library from URL
    remove     Remove library
    deploy     Find and add missing libraries
    publish    Publish program or library
    update     Update to branch, tag, revision or latest
    sync       Synchronize library references
    ls         View dependency tree
    status     Show version control status
               
    compile    Compile code using the mbed build tools
    test       Find, build and run tests
    export     Generate an IDE project
    detect     Detect connected mbed targets/boards
               
    config     Tool configuration
    target     Set or get default target
    toolchain  Set or get default toolchain
               
    help       This help screen
root@ubuntu:/usr/Ashok/mbed_CLI/blinky/oob-mbed-os-example-blinky# 
```

thomasonw, 2017-01-12:
I have the problem using native Ubuntu.  mbed command not recognized, using direct python call works.

```al@al-Precision-5510:~$ cat /etc/os-release
NAME="Ubuntu"
VERSION="16.04.1 LTS (Xenial Xerus)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 16.04.1 LTS"
VERSION_ID="16.04"
HOME_URL="http://www.ubuntu.com/"
SUPPORT_URL="http://help.ubuntu.com/"
BUG_REPORT_URL="http://bugs.launchpad.net/ubuntu/"
VERSION_CODENAME=xenial
UBUNTU_CODENAME=xenial
al@al-Precision-5510:~$ python --version
Python 2.7.12
al@al-Precision-5510:~$ pip --version
pip 8.1.1 from /usr/lib/python2.7/dist-packages (python 2.7)
al@al-Precision-5510:~$ mbed
No command 'mbed' found, did you mean:
 Command 'med' from package 'ncl-ncarg' (universe)
 Command 'mred' from package 'racket' (universe)
mbed: command not found
al@al-Precision-5510:~$ 
al@al-Precision-5510:~$ 
al@al-Precision-5510:~$ 
al@al-Precision-5510:~$ python ./.local/lib/python2.7/site-packages/mbed/mbed.py -h
usage: mbed [-h] [--version]             ...

Command-line code management tool for ARM mbed OS - http://www.mbed.com
version 1.0.0

Use 'mbed <command> -h|--help' for detailed help.
Online manual and guide available at https://github.com/ARMmbed/mbed-cli

optional arguments:
  -h, --help   show this help message and exit
  --version    print version number and exit

Commands:
             
    new        Create new mbed program or library
    import     Import program from URL
    add        Add library from URL
    remove     Remove library
    deploy     Find and add missing libraries
    publish    Publish program or library
    update     Update to branch, tag, revision or latest
    sync       Synchronize library references
    ls         View dependency tree
    status     Show version control status
               
    compile    Compile code using the mbed build tools
    test       Find, build and run tests
    export     Generate an IDE project
    detect     Detect connected mbed targets/boards
               
    config     Tool configuration
    target     Set or get default target
    toolchain  Set or get default toolchain
               
    help       This help screen
al@al-Precision-5510:~$ 
```

Any suggestion (Rather new to Linux...)

bridadan, 2017-01-12:
Hi @thomasonw can you post the following information to help us debug?

1. `echo $PATH`
2. `which python`
3. `which pip`

And can you try executing `mbed-cli` instead of `mbed`? I wouldn't expect you this to work, but I'd like to rule out the possibility.

Thanks!

thomasonw, 2017-01-12:
@bridadan Sure:
```
al@al-Precision-5510:~$ echo $PATH
/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin
al@al-Precision-5510:~$ python --version
Python 2.7.12
al@al-Precision-5510:~$ pip --version
pip 8.1.1 from /usr/lib/python2.7/dist-packages (python 2.7)
al@al-Precision-5510:~$ 
```

And these are the commands I used to install the software (IIRC):
```
sudo apt install gcc-arm-none-eabi
sudo apt install python-pip
pip install mbed-cli
```
Python seemed to already be installed as part of Ubunti

Thanks!

thomasonw, 2017-01-13:
@bridadan 
Sorry:   `mbed-cli` did not work, as you expected.  (Command not found)

bridadan, 2017-01-13:
@thomasonw I actually was hoping you could run the exact commands `which python` and `which pip`. That will tell me where your installed versions of python and pip live. I have a feeling that the place where your installation of pip is actually putting the installed packages is not in your path.

thomasonw, 2017-01-14:
@bridadn

 

Sure  (as I mentioned, new to linux --  still learning about commands, 'which' is a new one for me to look up..)

 

Here are the results:

`

al@al-Precision-5510:~$ which python

/usr/bin/python

al@al-Precision-5510:~$ which pip

/usr/bin/pip

al@al-Precision-5510:~$ 

`

thomasonw, 2017-01-16:
@bridadan Hello.   Just a bump, not sure the prior reply via Email notified you directly.

bridadan, 2017-01-17:
Hi @thomasonw, thanks for the bump, I got side tracked last time looking at this :smile: 

I think what's happening is when you run `pip install mbed-cli`, it's not placing the script in a directory that's included in your systems PATH.

For instance, this is the output of installing mbed-cli on a Linux machine I have:
```
$ pip install mbed-cli
Downloading/unpacking mbed-cli
  Downloading mbed-cli-1.0.0.zip (50kB): 50kB downloaded
  Running setup.py (path:/tmp/pip_build_root/mbed-cli/setup.py) egg_info for package mbed-cli

Installing collected packages: mbed-cli
  Running setup.py install for mbed-cli

    Installing mbed script to /usr/local/bin
    Installing mbed-cli script to /usr/local/bin
Successfully installed mbed-cli
Cleaning up...
```

The important line being `Installing mbed script to /usr/local/bin`.

And if I echo my `$PATH`:
```
$ echo $PATH
/home/bridan01/bin:/home/bridan01/.pyenv/shims:/home/bridan01/.pyenv/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games
```

We can see that `/usr/local/bin` is in the list, so it should be found just fine.

Can you please run the following commands and post the output?
```
pip uninstall mbed-cli
pip install mbed-cli
```

I'm curious to see where pip is installing the script.

kimphill, 2017-01-17:
Curiously, the pip uninstall and subsequent install made things work for my 16.10 setup:

```
$ pip uninstall mbed-cli
Uninstalling mbed-cli-0.9.7:
  /home/kim/.local/bin/mbed
  /home/kim/.local/bin/mbed-cli
  /home/kim/.local/lib/python2.7/site-packages/mbed/__init__.py
  /home/kim/.local/lib/python2.7/site-packages/mbed/__init__.pyc
  /home/kim/.local/lib/python2.7/site-packages/mbed/mbed.py
  /home/kim/.local/lib/python2.7/site-packages/mbed/mbed.pyc
  /home/kim/.local/lib/python2.7/site-packages/mbed_cli-0.9.7.dist-info/DESCRIPTION.rst
  /home/kim/.local/lib/python2.7/site-packages/mbed_cli-0.9.7.dist-info/INSTALLER
  /home/kim/.local/lib/python2.7/site-packages/mbed_cli-0.9.7.dist-info/METADATA
  /home/kim/.local/lib/python2.7/site-packages/mbed_cli-0.9.7.dist-info/RECORD
  /home/kim/.local/lib/python2.7/site-packages/mbed_cli-0.9.7.dist-info/WHEEL
  /home/kim/.local/lib/python2.7/site-packages/mbed_cli-0.9.7.dist-info/entry_points.txt
  /home/kim/.local/lib/python2.7/site-packages/mbed_cli-0.9.7.dist-info/metadata.json
  /home/kim/.local/lib/python2.7/site-packages/mbed_cli-0.9.7.dist-info/top_level.txt
Proceed (y/n)? y
  Successfully uninstalled mbed-cli-0.9.7
You are using pip version 8.1.2, however version 9.0.1 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
$ pip install mbed-cli
Collecting mbed-cli
  Downloading mbed-cli-1.0.0.zip (50kB)
    100% |████████████████████████████████| 51kB 1.3MB/s 
Building wheels for collected packages: mbed-cli
  Running setup.py bdist_wheel for mbed-cli ... done
  Stored in directory: /home/kim/.cache/pip/wheels/7c/18/81/ec1318c11501fc9ab8f39e8d50621137625461a17ab2928502
Successfully built mbed-cli
Installing collected packages: mbed-cli
Exception:
Traceback (most recent call last):
  File "/home/kim/.local/lib/python2.7/site-packages/pip/basecommand.py", line 215, in main
    status = self.run(options, args)
  File "/home/kim/.local/lib/python2.7/site-packages/pip/commands/install.py", line 317, in run
    prefix=options.prefix_path,
  File "/home/kim/.local/lib/python2.7/site-packages/pip/req/req_set.py", line 742, in install
    **kwargs
  File "/home/kim/.local/lib/python2.7/site-packages/pip/req/req_install.py", line 831, in install
    self.move_wheel_files(self.source_dir, root=root, prefix=prefix)
  File "/home/kim/.local/lib/python2.7/site-packages/pip/req/req_install.py", line 1032, in move_wheel_files
    isolated=self.isolated,
  File "/home/kim/.local/lib/python2.7/site-packages/pip/wheel.py", line 346, in move_wheel_files
    clobber(source, lib_dir, True)
  File "/home/kim/.local/lib/python2.7/site-packages/pip/wheel.py", line 317, in clobber
    ensure_dir(destdir)
  File "/home/kim/.local/lib/python2.7/site-packages/pip/utils/__init__.py", line 83, in ensure_dir
    os.makedirs(path)
  File "/usr/lib/python2.7/os.py", line 157, in makedirs
    mkdir(name, mode)
OSError: [Errno 13] Permission denied: '/usr/local/lib/python2.7/dist-packages/mbed_cli-1.0.0.dist-info'
You are using pip version 8.1.2, however version 9.0.1 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
kim@dupont ~$ sudo pip install mbed-cli
The directory '/home/kim/.cache/pip/http' or its parent directory is not owned by the current user and the cache has been disabled. Please check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
The directory '/home/kim/.cache/pip' or its parent directory is not owned by the current user and caching wheels has been disabled. check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
Collecting mbed-cli
Installing collected packages: mbed-cli
Successfully installed mbed-cli-1.0.0
You are using pip version 8.1.2, however version 9.0.1 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
$ which mbed
/usr/local/bin/mbed
$ mbed
usage: mbed [-h] [--version]             ...

Command-line code management tool for ARM mbed OS - http://www.mbed.com
version 1.0.0

Use 'mbed <command> -h|--help' for detailed help.
Online manual and guide available at https://github.com/ARMmbed/mbed-cli

optional arguments:
  -h, --help   show this help message and exit
  --version    print version number and exit

Commands:
             
    new        Create new mbed program or library
    import     Import program from URL
    add        Add library from URL
    remove     Remove library
    deploy     Find and add missing libraries
    publish    Publish program or library
    update     Update to branch, tag, revision or latest
    sync       Synchronize library references
    ls         View dependency tree
    status     Show version control status
               
    compile    Compile code using the mbed build tools
    test       Find, build and run tests
    export     Generate an IDE project
    detect     Detect connected mbed targets/boards
               
    config     Tool configuration
    target     Set or get default target
    toolchain  Set or get default toolchain
               
    help       This help screen
$ 

```

ansondtx20, 2017-01-17:
> 4bb","name":"GitHub"},"entity":{"external_key":"github/ARMmbed/mbed-cli","
> title":"ARMmbed/mbed-cli","subtitle":"GitHub
> repository","main_image_url":"https://cloud.githubusercontent.com/assets/1
> 43418/17495839/a5054eac-5d88-11e6-95fc-7290892c7bb5.png","avatar_image_url
> b-11e6-9aed-b52498112777.png","action":{"name":"Open in
> GitHub","url":"https://github.com/ARMmbed/mbed-cli"}},"updates":{"snippets
> ":[{"icon":"PERSON","message":"@kimphill in #364: Curiously, the pip
> setup:\r\n\r\n```\r\n$ pip uninstall mbed-cli\r\nUninstalling
> mbed-cli-0.9.7:\r\n  /home/kim/.local/bin/mbed\r\n
> /home/kim/.local/bin/mbed-cli\r\n
> /home/kim/.local/lib/python2.7/site-packages/mbed/__init__.py\r\n
> /home/kim/.local/lib/python2.7/site-packages/mbed/__init__.pyc\r\n
> /home/kim/.local/lib/python2.7/site-packages/mbed/mbed.py\r\n
> /home/kim/.local/lib/python2.7/site-packages/mbed/mbed.pyc\r\n
> RIPTION.rst\r\n
> ALLER\r\n
> DATA\r\n
> RD\r\n
> y_points.txt\r\n
> data.json\r\n
> level.txt\r\nProceed (y/n)? y\r\n  Successfully uninstalled
> mbed-cli-0.9.7\r\nYou are using pip version 8.1.2, however version 9.0.1
> is available.\r\nYou should consider upgrading via the 'pip install
> --upgrade pip' command.\r\n$ pip install mbed-cli\r\nCollecting
> mbed-cli\r\n  Downloading mbed-cli-1.0.0.zip (50kB)\r\n    100%
> [... 43 lines cut ...]
> configuration\r\n    target     Set or get default target\r\n
> toolchain  Set or get default toolchain\r\n               \r\n    help
>   This help screen\r\n$ \r\n\r\n```"}],"action":{"name":"View
> Issue","url":"https://github.com/ARMmbed/mbed-cli/issues/364#issuecomment-
> 273247041"}}}

Can you try this:

ubuntu% sudo pip uninstall mbed-cli
ubuntu% sudo pip install mbed-cli
ubuntu% which mbed

Thanks,

Doug

thomasonw, 2017-01-17:
Hello - thanks for keeping at this : -)

Here is the results of your suggestion:

```
al@al-Precision-5510:~$ pip uninstall mbed-cli
Uninstalling mbed-cli-1.0.0:
  /home/al/.local/bin/mbed
  /home/al/.local/bin/mbed-cli
  /home/al/.local/lib/python2.7/site-packages/mbed/__init__.py
  /home/al/.local/lib/python2.7/site-packages/mbed/__init__.pyc
  /home/al/.local/lib/python2.7/site-packages/mbed/mbed.py
  /home/al/.local/lib/python2.7/site-packages/mbed/mbed.pyc
  /home/al/.local/lib/python2.7/site-packages/mbed_cli-1.0.0.dist-info/DESCRIPTION.rst
  /home/al/.local/lib/python2.7/site-packages/mbed_cli-1.0.0.dist-info/INSTALLER
  /home/al/.local/lib/python2.7/site-packages/mbed_cli-1.0.0.dist-info/METADATA
  /home/al/.local/lib/python2.7/site-packages/mbed_cli-1.0.0.dist-info/RECORD
  /home/al/.local/lib/python2.7/site-packages/mbed_cli-1.0.0.dist-info/WHEEL
  /home/al/.local/lib/python2.7/site-packages/mbed_cli-1.0.0.dist-info/entry_points.txt
  /home/al/.local/lib/python2.7/site-packages/mbed_cli-1.0.0.dist-info/metadata.json
  /home/al/.local/lib/python2.7/site-packages/mbed_cli-1.0.0.dist-info/top_level.txt
Proceed (y/n)? y
  Successfully uninstalled mbed-cli-1.0.0
You are using pip version 8.1.1, however version 9.0.1 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
al@al-Precision-5510:~$ pip install mbed-cli
Collecting mbed-cli
Installing collected packages: mbed-cli
Successfully installed mbed-cli
You are using pip version 8.1.1, however version 9.0.1 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
al@al-Precision-5510:~$ 
```

And the results:

```
al@al-Precision-5510:~$ echo $PATH
/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin
al@al-Precision-5510:~$ which python
/usr/bin/python
al@al-Precision-5510:~$ which pip
/usr/bin/pip
al@al-Precision-5510:~$ 
al@al-Precision-5510:~$ mbed
No command 'mbed' found, did you mean:
 Command 'med' from package 'ncl-ncarg' (universe)
 Command 'mred' from package 'racket' (universe)
mbed: command not found
al@al-Precision-5510:~$ 
```
Still no joy.

In looking at the posts above, it seems I should have 'mbed' script in my /usr/local/bin folder, but I do not:

```
al@al-Precision-5510:~$ ls /usr/local/bin
actisense-serial  candump2analyzer        iptee                     n2kd          n2kd_nmea_monitor  request-group-function
analyzer          command-group-function  list-product-information  n2kd_monitor  nmea0183-serial    socketcan-writer
al@al-Precision-5510:~$ 
```
===========================================

I next tried things with the sudo, and we have success:

```
al@al-Precision-5510:~$ sudo pip uninstall mbed-cli
[sudo] password for al: 
The directory '/home/al/.cache/pip/http' or its parent directory is not owned by the current user and the cache has been disabled. Please check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
Uninstalling mbed-cli-1.0.0:
[... 59 lines cut ...]
               
    config     Tool configuration
    target     Set or get default target
    toolchain  Set or get default toolchain
               
    help       This help screen
al@al-Precision-5510:~$ 
```

Some more details:
```
al@al-Precision-5510:~$ which mbed
/usr/local/bin/mbed
al@al-Precision-5510:~$ ls /usr/local/bin
actisense-serial  candump2analyzer        iptee                     mbed      n2kd          n2kd_nmea_monitor  request-group-function
analyzer          command-group-function  list-product-information  mbed-cli  n2kd_monitor  nmea0183-serial    socketcan-writer
al@al-Precision-5510:~$ 

```
I
So, permissions was the issue and I needed to Sudo it.   Is that the intended procedure?  Though I seem to have things working now, if there is any other questions you have for my machine to help with the basic install, please do ask!

Thanks to all who chipped in.

-al-

ansondtx20, 2017-01-17:
> ":[{"icon":"PERSON","message":"@thomasonw in #364: Hello - thanks for
> keeping at this : -)\r\n\r\nHere is the results of your
> suggestion:\r\n\r\n```\r\nal@al-Precision-5510:~$ pip uninstall
> mbed-cli\r\nUninstalling mbed-cli-1.0.0:\r\n
> /home/al/.local/bin/mbed\r\n  /home/al/.local/bin/mbed-cli\r\n
> /home/al/.local/lib/python2.7/site-packages/mbed/__init__.py\r\n
> /home/al/.local/lib/python2.7/site-packages/mbed/__init__.pyc\r\n
> /home/al/.local/lib/python2.7/site-packages/mbed/mbed.py\r\n
> /home/al/.local/lib/python2.7/site-packages/mbed/mbed.pyc\r\n
> mbed-cli-1.0.0\r\nYou are using pip version 8.1.1, however version 9.0.1
> --upgrade pip' command.\r\nal@al-Precision-5510:~$ pip install
> mbed-cli\r\nCollecting mbed-cli\r\nInstalling collected packages:
> mbed-cli\r\nSuccessfully installed mbed-cli\r\nYou are using pip version
> 8.1.1, however version 9.0.1 is available.\r\nYou should consider
> command.\r\nal@al-Precision-5510:~$ \r\n```\r\n\r\nAnd the
> results:\r\n\r\n```\r\nal@al-Precision-5510:~$ echo
> $PATH\r\n/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr
> /games:/usr/local/games:/snap/bin\r\nal@al-Precision-5510:~$ which
> python\r\n/usr/bin/python\r\nal@al-Precision-5510:~$ which
> pip\r\n/usr/bin/pip\r\nal@al-Precision-5510:~$
> \r\nal@al-Precision-5510:~$ mbed\r\nNo command 'mbed' found, did you
> mean:\r\n Command 'med' from package 'ncl-ncarg' (universe)\r\n Command
> 'mred' from package 'racket' (universe)\r\nmbed: command not
> found\r\nal@al-Precision-5510:~$ \r\n```\r\nStill no joy.\r\n\r\nIn
> not:\r\n\r\n```\r\nal@al-Precision-5510:~$ ls
> [... 17 lines cut ...]
> mbed\r\n/usr/local/bin/mbed\r\nal@al-Precision-5510:~$ ls
> socketcan-writer\r\nal@al-Precision-5510:~$ \r\n\r\n```\r\nI\r\nSo,
> please do ask!\r\n\r\nThanks to all who chipped
> in.\r\n\r\n-al-\r\n"}],"action":{"name":"View
> 273271569"}}}

yep... typically, installing python packages with a global python install
(as is the case with MacOS and Ubuntu...), you typically need to "sudo"
things... 

Glad its working!  Enjoy and Happy New Year!

Doug

bridadan, 2017-01-17:
Many thanks to everyone involved here! Here's a full break down of @thomasonw's situation so if other people find this issue they can hopefully solve it themselves. This is not a fault of mbed-cli, but rather an issue with how Python and pip are set up on your system.

You can see from @thomasonw's log from the first installation that mbed-cli was installed to the path `/home/al/.local/bin/mbed`. The path `/home/al/.local/bin` is not in the system's PATH, so this is why the command was not recognized in the terminal.

By using `sudo`, it looks like pip switched the install directory to `/usr/local/bin`. Since that is in the system's PATH, the command was installed correctly.

If you don't want (or can't) use `sudo` to install python packages, you can just edit your PATH to include the first installation path. For @thomasonw, this would be `/home/al/.local/bin`. This is typically done by adding the following line to the files `~/.bashrc` or `~/.profile` (The `~` is shorthand for your home directory, in this case that's `/home/al`):

```
PATH="/home/al/.local/bin:$PATH"
```

This won't take affect until you re-login or until you run `source ~/.bashrc`.

Since there a number of different issues that have been discussed in this thread, I'm going to close this one for now and ask people to reopen separate issues if they need to be resolved.

closed 2017-01-17 by bridadan

thomasonw, 2017-01-17:
@bridadan 
Perhaps add something to the instructions about needing sudo?
[https://docs.mbed.com/docs/mbed-os-handbook/en/5.1/dev_tools/cli/]

Am sure for linux guys they saying must be "If at 1st you do not succeed, SUDO!" -- but for those coming in new, a hint might be helpful.

Thanks again to all for helping me on this.
-al-

bridadan, 2017-01-17:
@thomasonw There actually already is a hint there:

```
On Linux or Mac, you may need to run with sudo.
```

thomasonw, 2017-01-17:
@bridadan 

Well - so there is.   And with that I offer my apologies for not taking note of that before and spending folks time.   (RTFM)    Though am appreciative of the kind help here!

bridadan, 2017-01-17:
Not to worry! That's why we're here :+1: I was mostly just curious why it failed in the first place!

petertorelli, 2020-11-03:
The fact that `$HOME/.local/bin` needs to be in the user's path really needs to be added to the mbed CLI install page since it is guaranteed to happen if you are not running as root and aren't that familiar with Python CLI modules. [Version 6 docs here.](https://os.mbed.com/docs/mbed-os/v6.4/build-tools/install-and-set-up.html)