ARMmbed/mbed-cli #377: Can not build for DISCO_F051R8

zgoda, 2016-10-19:
Compilation fails:

```
$ mbed compile -t GCC_ARM -m DISCO_F051R8
Building project blinky-lib (DISCO_F051R8, GCC_ARM)
Scan: .
Scan: mbed
Scan: env
Compile: main.cpp
[ERROR] In file included from ./mbed/25aea2a3f4e3/mbed.h:30:0,
                 from ./main.cpp:1:
./mbed/25aea2a3f4e3/platform.h:19:20: fatal error: device.h: No such file or directory
 #include "device.h"
                    ^
compilation terminated.

[mbed] ERROR: "python" returned error code 1.
[mbed] ERROR: Command "python -u /home/jazg/tmp/mbed/blinky-lib/.temp/tools/make.py -t GCC_ARM -m DISCO_F051R8 --source . --build ./BUILD/DISCO_F051R8/GCC_ARM" in "/home/jazg/tmp/mbed/blinky-lib"
---
```

Project directory has been initialized as mbed 2.0:

```
$ mbed new blinky-lib --mbedlib
[mbed] Creating new program "blinky-lib" (git)
[mbed] Adding library "mbed" from "https://mbed.org/users/mbed_official/code/mbed/builds" at latest revision in the current branch
[mbed] Downloading mbed library build "25aea2a3f4e3" (might take a minute)
[mbed] Unpacking mbed library build "25aea2a3f4e3" in "/home/jazg/tmp/mbed/blinky-lib/mbed"
[mbed] Updating reference "mbed" -> "https://mbed.org/users/mbed_official/code/mbed/builds/25aea2a3f4e3"
[mbed] Couldn't find build tools in your program. Downloading the mbed 2.0 SDK tools...
```

This mcu is on the list of supported targets.

zgoda, 2016-10-19:
Building for mbed-os fails as well:

```

[...]

Link: blinky-os
./BUILD/DISCO_F051R8/GCC_ARM/mbed-os/rtos/rtx/TARGET_CORTEX_M/TARGET_M0/TOOLCHAIN_GCC/HAL_CM0.o: In function `SysTick_Handler':
HAL_CM0.S:(.text+0x12c): multiple definition of `SysTick_Handler'
./BUILD/DISCO_F051R8/GCC_ARM/mbed-os/hal/targets/cmsis/TARGET_STM/TARGET_STM32F0/TARGET_DISCO_F051R8/system_stm32f0xx.o:system_stm32f0xx.c:(.text.SysTick_Handler+0x0): first defined here
/usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/bin/ld: ./BUILD/DISCO_F051R8/GCC_ARM/blinky-os.elf section `.text' will not fit in region `FLASH'
/usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/bin/ld: region `FLASH' overflowed by 620 bytes
collect2: error: ld returned 1 exit status
[ERROR] ./BUILD/DISCO_F051R8/GCC_ARM/mbed-os/rtos/rtx/TARGET_CORTEX_M/TARGET_M0/TOOLCHAIN_GCC/HAL_CM0.o: In function `SysTick_Handler':
HAL_CM0.S:(.text+0x12c): multiple definition of `SysTick_Handler'
./BUILD/DISCO_F051R8/GCC_ARM/mbed-os/hal/targets/cmsis/TARGET_STM/TARGET_STM32F0/TARGET_DISCO_F051R8/system_stm32f0xx.o:system_stm32f0xx.c:(.text.SysTick_Handler+0x0): first defined here
/usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/bin/ld: ./BUILD/DISCO_F051R8/GCC_ARM/blinky-os.elf section `.text' will not fit in region `FLASH'
/usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/bin/ld: region `FLASH' overflowed by 620 bytes
collect2: error: ld returned 1 exit status

[mbed] ERROR: "python" returned error code 1.
[mbed] ERROR: Command "python -u /home/jazg/tmp/mbed/blinky-os/mbed-os/tools/make.py -t GCC_ARM -m DISCO_F051R8 --source . --build ./BUILD/DISCO_F051R8/GCC_ARM" in "/home/jazg/tmp/mbed/blinky-os"
---
```

This time project was initialized as mbed-os:

```
$ mbed new blinky-os
[mbed] Creating new program "blinky-os" (git)
[mbed] Adding library "mbed-os" from "https://github.com/ARMmbed/mbed-os" at latest revision in the current branch
[mbed] Updating reference "mbed-os" -> "https://github.com/ARMmbed/mbed-os/#9976738882d83a58b73d225594c5d7cfbff9d2ac"
```

zgoda, 2016-10-19:
For reference, here's the code that fails to compile:

```
$ cat main.cpp 
#include <mbed.h>

Serial pc(USBTX, USBRX);

int main(void) {
    pc.printf("Hello, world!\n");
    return 0;
}
```

screamerbg, 2016-10-19:
@zgoda Looks like ST Discovery L051R8 is not supported in mbed 2.0 releases - https://github.com/ARMmbed/mbed-os/blob/master/targets/targets.json#L1069 note that it doesn't have "release_versions" as with other targets.

Regarding mbed-os support, are you happy to file this issue against https://github.com/ARMmbed/mbed-os where members of the ST engineering team (@adustm @bcostm) might be able to help?

bcostm, 2016-10-19:
Hi,
The DISCO_F051C8 SDK is not officially supported by ST. But I don't think this is why the "release_versions" line is not present. Because I see this line for the NUCLEO_F767ZI or DISCO_F769NI and these boards are also not officially supported on mbed by ST. Maybe it is because no mbed 2 tests have been run on this platform.

So, what I suggest, is to launch all mbed 2 tests for this platform. If the tests are OK then send a PR to add the line `"release_versions": ["2"]`.

screamerbg, 2016-10-19:
@bcostm I'd rather like to ensure that this target is supported in mbed 5.0 than mbed 2.0. Unfortunately it doesn't have enough flash for it as it seems...

```
HAL_CM0.S:(.text+0x12c): multiple definition of `SysTick_Handler'
./BUILD/DISCO_F051R8/GCC_ARM/mbed-os/hal/targets/cmsis/TARGET_STM/TARGET_STM32F0/TARGET_DISCO_F051R8/system_stm32f0xx.o:system_stm32f0xx.c:(.text.SysTick_Handler+0x0): first defined here
c:/program files/gnu tools arm embedded/4.9 2015q2/bin/../lib/gcc/arm-none-eabi/4.9.3/../../../../arm-none-eabi/bin/ld.exe: ./BUILD/DISCO_F051R8/GCC_ARM/mbed-os-test.elf section `.text' will not fit in region `FLASH'
c:/program files/gnu tools arm embedded/4.9 2015q2/bin/../lib/gcc/arm-none-eabi/4.9.3/../../../../arm-none-eabi/bin/ld.exe: region `FLASH' overflowed by 540 bytes
```

zgoda, 2016-10-20:
This board has 48KB flash and I guess is not mbed-os capable.

bcostm, 2016-10-20:
Yes unfortunately there is not enough Flash. We need more than 64kb...

wassfila, 2017-08-13:
Just as a hint, consider using "--specs=nano.specs", I have mbed os 5 compiled for around 35 KB so just give it a try if you want, example with my own branched os [here ](https://github.com/wassfila/STM32_IoT_HelloWorld/blob/master/08_mbed_cli_os_BluePill_Target/develop_nano.json)

closed 2018-05-26 by screamerbg