ARMmbed/mbed-cli #378: mbed deploy does not respect .mbedignore

bremoran, 2016-10-19:
I have a repo that contains several directories, including a PCB design. The PCB design contains files named `<library name>.lib`. I have added a `.mbedignore` file, which specifies all the non-code directories, but `mbed deploy` does not respect this, insisting on trying to parse my PCB library directories.

teetak01, 2016-10-24:
Same here. I tried ignoring both the `<library name>.lib` and `<library name>` folder, but `mbed deploy` still very persistently likes to do useless deploys.

bremoran, 2016-10-24:
@teetak01 Apparently, `mbed deploy`'s lack of respect for `.mbedignore` is by design.

teetak01, 2016-10-25:
@screamerbg @theotherjimmy any comments on this? I am just wondering what the benefit is for the user if `.mbedignore` is not supposed to affect `mbed deploy`. Or could this functionality be reconsidered?

At least on this repository (or https://docs.mbed.com/docs/mbed-os-handbook/en/5.1/dev_tools/cli/) I do not see really much documentation on how `.mbedignore `is supposed to work, other than some reference that `mbed update` at least works with it.

bridadan, 2016-10-25:
@teetak01 I might be able to supply some background as to why `.mbedignore` files aren't currently supported in this context.

The simple reason is that the logic that respects the `.mbedignore` files is located in `mbed-os` at the moment. It was originally only used during the compilation process. There other factors that are factored into the global "ignore logic" besides the `.mbedignore` files, so this is all consolidated within those tools.

teetak01, 2016-10-28:
@bridadan thank you for your comments. Has there been any thought on refactoring the tools away from mbed-os? Keep the OS as its own entity and concentrate the tools either here or in third repository (if there is other dependencies). Currently there is maybe a bit too tight coupling between mbed-os and mbed-cli versions and these two cannot deviate too far before strange errors start to occur, possible because some of the tool changes are done on both sides. But yes, I can image this is not trivial in any case.

bridadan, 2016-10-28:
@teetak01 There's definitely been talk about this. And I don't know the exact time frame on any of those changes (if they were to happen at all), specifically because they are non-trivial like you said. Right now the source code and tools are fairly tightly coupled. Though in the future, especially as development smooths out, I could see this happening when there is less feature development on the tools.

bmcdonnell-ionx, 2017-12-14:
> Apparently, `mbed deploy`'s lack of respect for `.mbedignore` is by design.

Why should it be that way?

You wind up telling one build system (`mbed build`) to ignore the files, but all the others (via `mbed export`) not to. (Right?) If you make any customizations to your exported project files to ignore the same things there, those modifications will be overwritten next time you `mbed export`.

theotherjimmy, 2017-12-14:
@bmcdonnell-ionx `mbed export` heeds `.mbedignore` files.

bmcdonnell-ionx, 2017-12-14:
> `mbed export` heeds `.mbedignore` files.

So this issue has been addressed, and should be closed?

bridadan, 2017-12-14:
Well the issue is specifically that `mbed deploy` still deploys `.lib` files that are either ignored directly by `.mbedignore` files or they are contained within a directory that is ignore by `.mbedignore` files.

It sounds like we don't intend to change this, so I think it's still safe to close this as "won't fix" @theotherjimmy ?

theotherjimmy, 2017-12-19:
[The documentation](https://os.mbed.com/docs/v5.6/tools/ignoring-files-from-mbed-build.html) needs to be updated to explain that it affects `mbed export` as well as `mbed compile`. At the moment, it references the non-existent `mbed build` command.

bmcdonnell-ionx, 2018-01-31:
How can I ignore specific files in an `mbed export`ed project? And why is `mbed export` overwriting `.mbedignore`?

I have a project that uses a library, where the library has a `test.cpp` file in its directory. I can't build the Mbed project with that file.

When I add `libname/test.cpp` to my `.mbedignore` file, `.mbedignore` gets overwritten by `mbed export`. But there is apparently *some* way to ignore things in exported projects, as e.g. half the folders under `mbed-os/features/` and all but one `TARGET`s are excluded from my exported project build.

theotherjimmy, 2018-01-31:
@bmcdonnell-ionx  
>And why is mbed export overwriting .mbedignore?

Certain exporters may write their own `.mbedignore` files.

> How can I ignore specific files in an `mbed export`ed project?

The `.mbedignore` files will work for both `mbed compile` and `mbed export`. I'm a little confused on what you mean do you mean `.mbedignore` then `mbed export`? or `mbed export` then (some IDE interaction) then `.mbedignore`? For either, you could just write in the `.mbedignore` file.

> When I add libname/test.cpp to my .mbedignore file, .mbedignore gets overwritten by mbed export.

You could put the `.mbedignore` file into `libname` with the contents:
```
test.cpp
```
And that would do it.

bmcdonnell-ionx, 2018-01-31:
> Certain exporters may write their own `.mbedignore` files.

Bummer.

I could see them adding to it, but clobbering it seems like bad behavior, IMO.

> I'm a little confused on what you mean do you mean `.mbedignore` then `mbed export`? or `mbed export` then (some IDE interaction) then `.mbedignore`? For either, you could just write in the `.mbedignore` file.

Step 1. I added `libname/test.cpp` to the `.mbedignore` file.
Step 2. `mbed export -m LPC4088 -i mcuxpresso`
Step 3. Inspect `.mbedignore`. It has been overwritten, and no longer contains `libname/test.cpp`.

> You could put the `.mbedignore` file into `libname`

I don't want to pollute a library that knows nothing of Mbed.

theotherjimmy, 2018-01-31:
@bmcdonnell-ionx This sounds like an issue with `mcuxpresso`. I think mcuxpresso may not support .mbedignore of single files. Let's discuss this on another issue in the mbed-os repo, I think we have ventured off topic of this issue.

bmcdonnell-ionx, 2018-01-31:
> I think we have ventured off topic of this issue.

Agreed. See reference above.

closed 2018-06-29 by AnotherButler