ARMmbed/mbed-cli #380: Incorrect LIBRARIES variable in exported Makefile

sdive, 2016-10-20:
Hi.

`mbed export -i eclipse_gcc_arm` produces an incorrect makefile:

```MakeFile
LIBRARY_PATHS := -L.././mbed-os/features/FEATURE_UVISOR/targets/TARGET_UVISOR_SUPPORTED/TARGET_MAX32550/TARGET_RELEASE/TARGET_M3 
LIBRARIES := -lbed-os/features/FEATURE_UVISOR/targets/TARGET_UVISOR_SUPPORTED/TARGET_MAX32550/TARGET_RELEASE/TARGET_M3/libconfiguration_max32550_cortex_m3 
```

It should be:
```MakeFile
LIBRARY_PATHS := -L.././mbed-os/features/FEATURE_UVISOR/targets/TARGET_UVISOR_SUPPORTED/TARGET_MAX32550/TARGET_RELEASE/TARGET_M3 
LIBRARIES := -lconfiguration_max32550_cortex_m3 
```

The project config is 
```
ROOT=.
TARGET=MAX32550
TOOLCHAIN=GCC_ARM
```

I fixed the issue by the following change:

```Diff
diff --git a/tools/export/makefile/__init__.py b/tools/export/makefile/__init__.py
index d29b273..d1731fc 100644
--- a/tools/export/makefile/__init__.py
+++ b/tools/export/makefile/__init__.py
@@ -44,7 +44,7 @@ class Makefile(Exporter):
                           self.resources.c_sources +
                           self.resources.cpp_sources]
 
-        libraries = [splitext(lib)[0][3:] for lib in self.resources.libraries]
+        libraries = [splitext(basename(lib))[0][3:] for lib in self.resources.libraries]
 
         ctx = {
             'name': self.project_name,
```

NB: The project builds correctly using `mbed compile`.

screamerbg, 2016-10-20:
CC @sarahmarshy @theotherjimmy

theotherjimmy, 2016-10-20:
This change is already in master

theotherjimmy, 2016-10-20:
https://github.com/ARMmbed/mbed-os/pull/3061 was merged, it contains a cross-toolchain compatible version of what you mentioned above @sdive

theotherjimmy, 2017-11-28:
Closing as fixed in mbed-os

closed 2017-11-28 by theotherjimmy