ARMmbed/mbed-cli #382: mbed-cli creates extra folders "none"

teetak01, 2016-10-25:
I have very strange issue with mbed-cli. It creates extra folders named `none` when ever `mbed` commands are used.

```
For. ex. 
mkdir test
cd test
mbed new .
ls -ltr
drwxrwxr-x  3 teetak01 teetak01 4096 loka  25 09:34 none
-rw-rw-r--  1 teetak01 teetak01   77 loka  25 09:34 mbed-os.lib
drwxrwxr-x 15 teetak01 teetak01 4096 loka  25 09:34 mbed-os
-rw-rw-r--  1 teetak01 teetak01 1329 loka  25 09:34 mbed_settings.py
```

The folder `none` contains:

```
8,0K	./github.com/ARMmbed/mbed-os/.git/info
4,0K	./github.com/ARMmbed/mbed-os/.git/refs/tags
8,0K	./github.com/ARMmbed/mbed-os/.git/refs/heads
8,0K	./github.com/ARMmbed/mbed-os/.git/refs/remotes/origin
12K	./github.com/ARMmbed/mbed-os/.git/refs/remotes
28K	./github.com/ARMmbed/mbed-os/.git/refs
4,0K	./github.com/ARMmbed/mbed-os/.git/objects/info
86M	./github.com/ARMmbed/mbed-os/.git/objects/pack
86M	./github.com/ARMmbed/mbed-os/.git/objects
4,0K	./github.com/ARMmbed/mbed-os/.git/branches
44K	./github.com/ARMmbed/mbed-os/.git/hooks
8,0K	./github.com/ARMmbed/mbed-os/.git/logs/refs/heads
8,0K	./github.com/ARMmbed/mbed-os/.git/logs/refs/remotes/origin
12K	./github.com/ARMmbed/mbed-os/.git/logs/refs/remotes
24K	./github.com/ARMmbed/mbed-os/.git/logs/refs
32K	./github.com/ARMmbed/mbed-os/.git/logs
87M	./github.com/ARMmbed/mbed-os/.git
87M	./github.com/ARMmbed/mbed-os
87M	./github.com/ARMmbed
87M	./github.com
87M	.
```
In some cases I have also seen some linker-problems arising from this when it `deploys `multiple copies of mbed-os due to `none`.

My mbed-cli version is 0.9.9
Using Ubuntu 16.04

Pip list:
```
adium-theme-ubuntu (0.3.4)
argcomplete (1.4.1)
beautifulsoup4 (4.5.1)
certifi (2016.2.28)
cffi (1.7.0)
chardet (2.3.0)
CoAPthon (2.0.0)
colorama (0.3.7)
coloredlogs (5.0)
coverage (4.2)
cryptography (1.4)
docutils (0.12)
enum34 (1.1.6)
fasteners (0.14.1)
funcsigs (1.0.2)
functools32 (3.2.3.post2)
futures (2.1.6)
fuzzywuzzy (0.12.0)
gcovr (3.2)
hgapi (1.7.2)
html5lib (0.999)
humanfriendly (1.44.7)
idna (2.1)
intelhex (2.1)
ipaddress (1.0.16)
Jinja2 (2.8)
jsonmerge (1.2.1)
jsonpointer (1.10)
jsonschema (2.4.0)
junit-xml (1.7)
lockfile (0.12.2)
Logbook (1.0.0)
lxml (3.5.0)
MarkupSafe (0.23)
mbed-cli (0.9.9)
mbed-clitest (0.6.0)
mbed-flasher (0.4.0)
mbed-greentea (1.2.2)
mbed-host-tests (1.1.3)
mbed-ls (1.2.8)
mbed-test-wrapper (1.0.0)
mdsrestclient (0.1.0)
meld (3.14.2)
mercurial (3.7.3)
mock (2.0.0)
monotonic (1.2)
ndg-httpsclient (0.4.2)
netifaces (0.10.5)
pathlib (1.0.1)
pbr (1.10.0)
pip (8.1.2)
prettytable (0.7.2)
project-generator (0.9.9)
project-generator-definitions (0.2.28)
psutil (4.3.0)
py (1.4.31)
pyasn1 (0.1.9)
pycparser (2.14)
pycurl (7.43.0)
pyelftools (0.23)
PyGithub (1.26.0)
Pygments (2.1.3)
pygobject (3.20.0)
PyJWT (1.4.1)
pyOCD (0.7.1)
pyOpenSSL (16.0.0)
pyserial (3.2.1)
pyshark (0.3.6.1)
pyusb (1.0.0)
pyxdg (0.25)
PyYAML (3.12)
raas-client (0.5.0)
requests (2.11.1)
semantic-version (2.5.0)
semver (2.6.0)
setuptools (20.7.0)
six (1.10.0)
socketIO-client (0.6.5)
Sphinx (1.2.2)
trollius (1.0.4)
Twisted (16.3.0)
unity-lens-photos (1.0)
valinor (0.0.15)
websocket-client (0.37.0)
xmltodict (0.10.2)
yattag (1.6.0)
yotta (0.16.4)
zope.interface (4.2.0)
```
mbed-cli config:

```
[mbed] Global config:
ARM_PATH=/home/teetak01/ARM_Compiler_5.06u3
TOOLCHAIN=GCC_ARM
CACHE=none
PROTOCOL=ssh
TARGET=K64F

[mbed] Local config (/home/teetak01/devel/test):
No local configuration is set
```
Any idea what might be causing this?

screamerbg, 2016-10-25:
@teetak01 Thanks for reporting this. I'll investigate and let you know soon. In the meantime can you update mbed CLI to latest (0.9.10) and try again?

teetak01, 2016-10-25:
@screamerbg updated to mbed-cli (0.9.10).

Same issue still. I first clone the mbed-os, and then possibly during `syncing?` it creates that extra folder?
```

teetak01@machine:~/devel/test2$ mbed new .
[mbed] Creating new program "test2" (git)
[mbed] Adding library "mbed-os" from "ssh://git@github.com/ARMmbed/mbed-os.git" at latest revision in the current branch
[mbed] Updating reference "mbed-os" -> "https://github.com/ARMmbed/mbed-os/#e2efb35723884fd21957c5fb1b9f4bdb33bd5cfa"
teetak01@machine:~/devel/test2$ ls -ltr
total 16
drwxrwxr-x  3 teetak01 teetak01 4096 loka  25 11:33 none
-rw-rw-r--  1 teetak01 teetak01   77 loka  25 11:33 mbed-os.lib
drwxrwxr-x 15 teetak01 teetak01 4096 loka  25 11:33 mbed-os
-rw-rw-r--  1 teetak01 teetak01 1329 loka  25 11:33 mbed_settings.py
```

teetak01, 2016-10-27:
The root-cause seems to be that I had defined the `cache=none` which the started to create caching folders in multiple places? Removing the cache definition completely (`mbed config -G cache --unset`) solved the issue for me. Thanks for debug help @screamerbg

screamerbg, 2016-10-28:
Fix coming with #356

screamerbg, 2016-11-25:
Fixed in latest mbed CLI (v1.0.0)

closed 2016-11-25 by screamerbg