ARMmbed/mbed-cli #384: Crash running `mbed new`

Sentimentron, 2016-10-28:
Following the instructions at https://docs.mbed.com/docs/mbed-os-handbook/en/5.1/dev_tools/cli/, installed the mbed command line tools via pip. 

Running `mbed new mbed-blinky` results in the following error:

`Traceback (most recent call last):
  File "/usr/local/bin/mbed", line 9, in <module>
    load_entry_point('mbed-cli==0.9.10', 'console_scripts', 'mbed')()
  File "/usr/local/lib/python3.4/site-packages/setuptools-15.2-py3.4.egg/pkg_resources/__init__.py", line 552, in load_entry_point
  File "/usr/local/lib/python3.4/site-packages/setuptools-15.2-py3.4.egg/pkg_resources/__init__.py", line 2672, in load_entry_point
  File "/usr/local/lib/python3.4/site-packages/setuptools-15.2-py3.4.egg/pkg_resources/__init__.py", line 2345, in load
  File "/usr/local/lib/python3.4/site-packages/setuptools-15.2-py3.4.egg/pkg_resources/__init__.py", line 2351, in resolve
  File "/usr/local/lib/python3.4/site-packages/mbed/mbed.py", line 989
    sorted_scms = sorted(sorted_scms, key=lambda (m, _): not m)
                                                 ^
SyntaxError: invalid syntax`

Environment is Mac OS X El Capitan, 10.11.6. Python version is 2.7.9 installed via Homebrew, Python 3 version is 3.7.9, again installed via Homebrew. Let me know if you need any more details.

bridadan, 2016-10-28:
@Sentimentron I think it's using your Python 3 installation, however mbed CLI only works with Python 2 as described here: https://github.com/ARMmbed/mbed-cli#requirements

Can you set your default Python installation to Python 2? By that I mean please be sure that the command `python` uses Python 2 instead of Python 3.

Sentimentron, 2016-10-29:
Ah, OK after some investigation, I found out that the `pip` command on my system comes from Python 3, rather than Python 2. It installs fine with Python 3's pip, but doesn't actually run. I now have a different issue about linking with OpenSSL with Python 2.

closed 2016-10-29 by Sentimentron