ARMmbed/mbed-cli #385: mbed compile (linux,windows)

orcim, 2016-10-29:
Hi

Why with different O.S. but with the same enviroments, i have different results.

Windows
=======
| Module    | .text | .data | .bss |
| Subtotals | 19744 |   136 |  620 |

Linux
====
| Module    | .text | .data | .bss |
| Subtotals | 37144 |  2192 |  664 |

the enviroments of  my trys

try in Windows
==============

Microsoft Windows [Versione 6.1.7601]
Copyright (c) 2009 Microsoft Corporation. Tutti i diritti riservati.

C:\Lavoro\_loris\_installed\_work\mbed_blinky>mbed config --list
[mbed] Global config:
No global configuration is set

[mbed] Local config (C:\Lavoro\_loris\_installed\_work\mbed_blinky):
TOOLCHAIN=GCC_ARM
TARGET=NUCLEO_L053R8

C:\Lavoro\_loris\_installed\_work\mbed_blinky>mbed compile
Building project mbed_blinky (NUCLEO_L053R8, GCC_ARM)
Scan: .
Scan: mbed
Scan: env
+-----------+-------+-------+------+
| Module    | .text | .data | .bss |
+-----------+-------+-------+------+
| Subtotals | 19744 |   136 |  620 |
+-----------+-------+-------+------+
Allocated Heap: unknown
Allocated Stack: unknown
Total Static RAM memory (data + bss): 756 bytes
Total RAM memory (data + bss + heap + stack): 756 bytes
Total Flash memory (text + data + misc): 19880 bytes
Image: .\BUILD\NUCLEO_L053R8\GCC_ARM\mbed_blinky.bin

C:\Lavoro\_loris\_installed\_work\mbed_blinky>arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 5.4.1 20160919 (releas
e) [ARM/embedded-5-branch revision 240496]
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

C:\Lavoro\_loris\_installed\_work\mbed_blinky>

try in Linux
============

orcim@orcim-UX32VD ~/0work/_myTry/_env/_mBed/__work/mbed_blinky $ mbed config --list
[mbed] Global config:
No global configuration is set

[mbed] Local config (/home/orcim/0work/_myTry/_env/_mBed/__work/mbed_blinky):
TOOLCHAIN=GCC_ARM
TARGET=NUCLEO_L053R8

orcim@orcim-UX32VD ~/0work/_myTry/_env/_mBed/__work/mbed_blinky $ mbed compile
Building project mbed_blinky (NUCLEO_L053R8, GCC_ARM)
Scan: .
Scan: mbed
Scan: env
+-----------+-------+-------+------+
| Module    | .text | .data | .bss |
+-----------+-------+-------+------+
| Subtotals | 37144 |  2192 |  664 |
+-----------+-------+-------+------+
Allocated Heap: unknown
Allocated Stack: unknown
Total Static RAM memory (data + bss): 2856 bytes
Total RAM memory (data + bss + heap + stack): 2856 bytes
Total Flash memory (text + data + misc): 39336 bytes
Image: ./BUILD/NUCLEO_L053R8/GCC_ARM/mbed_blinky.bin
orcim@orcim-UX32VD ~/0work/_myTry/_env/_mBed/__work/mbed_blinky $ arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 5.4.1 20160919 (release) [ARM/embedded-5-branch revision 240496]
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

orcim@orcim-UX32VD ~/0work/_myTry/_env/_mBed/__work/mbed_blinky $

screamerbg, 2016-11-25:
cc @theotherjimmy @bridadan

theotherjimmy, 2017-11-28:
This issue was related to link order and has been closed for quite some time in mbed-os. The Linux and windows builds have the same size, across all supported compilers.

closed 2017-11-28 by theotherjimmy