ARMmbed/mbed-cli #386: mbed-cli should be compatible with Coverity

miklis, 2016-11-03:
Trying to run `cov-build mbed compile -t GCC_ARM -m K64F` on a test application does not produce any results presumably due to Coverity being unable to read mbed-cli output even though we have Coverity configured to read output from both arm-none-eabi-gcc and armcc. 

Received message from the run is `[WARNING] No files were emitted. This may be due to a problem with your configurationor because no files were actually compiled by your build command.`

jankii01, 2016-11-03:
This is a quite serious blocking issue, needs to get fixed ASAP.

miklis, 2016-11-03:
Actually seems to work, just needed to reconfigure the compilers with new flags for Coverity.

closed 2016-11-03 by miklis

screamerbg, 2016-11-03:
@miklis @jankii01 can you file this against https://github.com/ARMmbed/mbed-os as the very core formatting logic is handled in the build system core?