ARMmbed/mbed-cli #387: Should be a link to information regarding profiles

adbridge, 2016-11-04:
There is a command line option to specify a profile file. There is no information however about the profiles. There should at least be a comment along the lines of : 
"For more information about profiles , see . . . . . " and point to the docs in mbed-os

adbridge, 2016-11-04:
https://github.com/ARMmbed/mbed-os/blob/master/docs/Toolchain_Profiles.md

bridadan, 2016-11-04:
It might be too long to include the link from above, but you could say something along the lines of "Please see the mbed OS documentation for more details" on the command line, and provide a link to the actual doc in the mbed-cli README.

If that makes sense to you, would you be able to add that?

screamerbg, 2018-05-26:
Now implemented. Thanks @adbridge !

closed 2018-05-26 by screamerbg