ARMmbed/mbed-cli #389: Exporting to Simplicity Studio produces unimportable project

kubaraczkowski, 2016-11-08:
Hi,

Please point me to a better location for complaining about the Simplicity Studio exporter if this is not OK here.

After cloning an mbed project I did `mbed export -i simplicityv3` which produced .slsproj file and so on, but importing that gives : "Path must include project and resource name: /Unnamed_Project"

Any clue about that?
Thanks in advance!

screamerbg, 2016-11-10:
cc @theotherjimmy

theotherjimmy, 2016-11-10:
This should be an issue on https://github.com/armmbed/mbed-os.

Further, what project are you exporting and what version of mbed-os are you using?

theotherjimmy, 2016-11-10:
I was able to reproduce the problem with the target EFM32GG_STK3700 with the mbed-os-example-blinky

screamerbg, 2018-05-26:
@kubaraczkowski Is this still an issue with Mbed OS 5.8? @stevew817

kubaraczkowski, 2018-05-31:
I can't comment on this anymore I'm afraid...

bmcdonnell-ionx, 2019-01-23:
I just tried

```
mbed import mbed-os-example-blinky
cd mbed-os-example-blinky
mbed export -i eclipse_gcc_arm -m EFM32GG11_STK3701
```

It produces `.cproject` and `.project` files, but Simplicity Studio v4 doesn't see it to import. (I right click in Project Explorer, Import &rarr; MCU Project, browse to the `mbed-os-example-blinky` folder, but it doesn't detect the project.)

Mbed CLI version 1.8.3.
https://github.com/ARMmbed/mbed-os-example-blinky/commit/86294c2a9681aa7a75aa9e574e0308b190fa16f5

theotherjimmy, 2019-01-24:
Hi @bmcdonnell-ionx,

I'm sorry that that specific exporter did not work for simplicity. Perhaps you could try `gnuarmeclipse` as well? 

-------------------

Note that neither of these exporters are designed to produce correct Simplicity Studio project files and that they very well may not .

bmcdonnell-ionx, 2019-01-24:
@theotherjimmy,

> Perhaps you could try `gnuarmeclipse` as well?

Same result.

stevew817, 2019-01-24:
the simplicity studio exporter was supposed to be deprecated and removed?

theotherjimmy, 2019-01-24:
@stevew817 I think so yes.

----------

@bmcdonnell-ionx Sorry we don't have an exporter that imports into simplicity for you. You could generate a `make_gcc_arm` export, and add a "makefile based project" if simplicity has the option. You could look for the includes within the makefile for assistance in teaching simplicity about the project.

bmcdonnell-ionx, 2019-01-24:
>> the simplicity studio exporter was supposed to be deprecated and removed?
> ... I think so yes.

@theotherjimmy, why?

theotherjimmy, 2019-01-25:
@bmcdonnell-ionx In general exporters are a huge support burden, as they introduce another axis along which things can break. This puts us up to 3 major variables:
 1) Mbed OS Version (5.1.x ... 5.11.x, Mbed OS 2 v50 ... v160)
 2) Target (I think we're in the 200s)
 3) Exporter (We have something like 15 of them)

In general, I maintain the exporters. I don't use IDEs (I'm a vi/vim/nvim/kakoune user), so I'm unlikely to notice odd issues or paper cuts. Further, we have limited CI resources and time. That means we have to be selective with our testing. We can't easily limit axis 1 or 2, so we could limit axis 3 to help keep coverage of exporters up. 

-------------

Also, it was an exporter for Simplicity v3. My understanding is that v3 is out of date/deprecated.

bmcdonnell-ionx, 2019-01-25:
@theotherjimmy, thanks for the insight.

FWIW, I think the exporters are part of the appeal of Mbed OS, as they can provide a fairly smooth path (when they work) to getting up and running with an IDE, including debugging.

Would having some shared core code for the Eclipse-based IDE exporters help to reduce your maintenance?

theotherjimmy, 2019-01-25:
> I think the exporters are part of the appeal of Mbed OS, as they can provide a fairly smooth path (when they work) to getting up and running with an IDE, including debugging.

I don't think it's a surprise that I'm not of the same opinion. Would having a project file that uses Mbed CLI have the same benefits? I wonder what we would call that.

> Would having some shared core code for the Eclipse-based IDE exporters help to reduce your maintenance?

Yes. I appreciate the thought as well.

bmcdonnell-ionx, 2019-01-25:
What should be done with this issue, then? Move to Mbed OS? (As [noted by @theotherjimmy above](https://github.com/ARMmbed/mbed-cli/issues/389#issuecomment-259806064).)

I'd like for my team to have the option to use Mbed OS on future projects. A Simplicity Studio exporter would probably facilitate that.

theotherjimmy, 2019-01-25:
> I'd like for my team to have the option to use Mbed OS on future projects. A Simplicity Studio exporter would probably facilitate that.

@JoeTheGuitarist @stevew817 @ChiefBureaucraticOfficer @SenRamakri  FYI

star297, 2019-02-24:
Ironic this, I have been away from this for a couple of years, just got back again and was going to use Mbed and Simplicity Studio because of the Energy Profiler.
You should remove other exporters, Simplicity Studio is the only one I know of that has an Energy Profiler.
This will result in Mbed is loosing some of its Low Power programming capability, you can't build a house without a tape measure, now it will be very difficult to accurately MCU processes. 
But if no one is using/supporting it then there really is no point it has to go (along with my Si-Labs boards to Ebay).

40Grit, 2019-02-24:
Great discussion here.

My team specifically ditched the exporters because we did not want to re-export everytime we wanted to make a project change and we aren't fans of using make.

We determined that VS Code had come along far enough, thanks to the "Language Server" and "debug adapter protocol".

We've created a simple enough project template that we can start up a new project quickly and no longer need to export a project to vs-code even.

While I once was a consumer of the exporters and still understand their value, I now would be completely unaffected if @theotherjimmy's team decided to work on something of higher value.

Regarding power point profiling. I highly recommend picking up a DAQ or other data logging solution. I have a team of 2 sometimes 3 and we've been able to get everything I've mentioned and more implemented in a short period of time.

I highly recommend freeing yourselves from vendor specific tools and eclipse environments. It's a liberating experience.

star297, 2019-02-24:
I agree, that's exactly what I was finding. I did try VS a couple of years back, nice layout, but went back to Mbed on-line, I just prefer the Keil based compiler and I can get on with uVision. I think we all have our preferences. May have to bite the bullet and buy a Keil licence :(

stevew817, 2019-02-24:
@star297 No need for such drastic measures... You can absolutely energy profile an application that was built outside of Simplicity Studio. I do it all the time.

I'll update this with a little how-to when I get to my dev machine.

40Grit, 2019-02-24:
I have comments on this elsewhere but the goal is to do power profiling, testing, debug, all through remote device farms. 

Eventually we will eliminate the "dev machine" concept completely by moving to eclipse Che. Best part of our setup... All free software tools.

mark-edgeworth, 2019-06-12:
Hi all, thanks for all the comments. The support for simplicityv3 looks to have been removed some time ago so I'm going to close this issue now. Do take a look at Arm's Mbed Studio as this might be closer to what you need in an IDE.

closed 2019-06-12 by mark-edgeworth