ARMmbed/mbed-cli #390: mbed import command: error to download mbed library

bcostm, 2016-11-10:
When I use this command:
`mbed import https://developer.mbed.org/teams/ST/code/Nucleo_blink_led/`

The mbed library cannot be downloaded:
```
[mbed] Importing program "Nucleo_blink_led" from "https://developer.mbed.org/teams/ST/code/Nucleo_blink_led" at latest revision in the current branch
[mbed] Adding library "mbed" from "http://mbed.org/users/mbed_official/code/mbed/builds" at rev #0ab6a29f35bf
[mbed] Downloading mbed library build "0ab6a29f35bf" (might take a minute)
[mbed] ERROR: Download failed!
[mbed] ERROR: Please try again later.
```

If I use another example containing other libraries like this one:
`mbed import https://developer.mbed.org/teams/ST/code/DISCO-F746NG_LCD_demo/`

All other libraries are well downloaded but not the mbed library:
```
[mbed] Importing program "DISCO-F746NG_LCD_demo" from "https://developer.mbed.org/teams/ST/code/DISCO-F746NG_LCD_demo" at latest revision in the current branch
[mbed] Adding library "BSP_DISCO_F746NG" from "https://developer.mbed.org/teams/ST/code/BSP_DISCO_F746NG" at rev #fe313c53cdb5
[mbed] Adding library "LCD_DISCO_F746NG" from "https://developer.mbed.org/teams/ST/code/LCD_DISCO_F746NG" at rev #d44525b1de98
[mbed] Adding library "mbed" from "http://mbed.org/users/mbed_official/code/mbed/builds" at rev #0ab6a29f35bf
[mbed] Downloading mbed library build "0ab6a29f35bf" (might take a minute)
[mbed] ERROR: Download failed!
[mbed] ERROR: Please try again later.
```

I have modified the mbed.bld file locally in order to download the `mbed-dev`  instead:
`http://mbed.org/users/mbed_official/code/mbed-dev/`
Then I have launched `mbed deploy`. It is very long but it works.

I don't know if I'm doing something wrong or if there is a real issue ?

screamerbg, 2017-01-11:
@bcostm I've actually tried reproducing the issue just after you reported it and it seems that it was related to issues with the mbed website and the ability to download releases. Could you please try to reproduce it again?

screamerbg, 2017-01-11:
cc @immunda

screamerbg, 2017-03-23:
Please reopen if reoccurs

closed 2017-03-23 by screamerbg