ARMmbed/mbed-cli #391: [ERROR] Attribute 'default_build' not found in target 'LPC1768'

sillevl, 2016-11-10:
Hi, I am using Ubuntu Mate 16.04. 
I installed mbed-cli using pip, and installed GCC ARM Embedded from https://launchpad.net/gcc-arm-embedded 

I added the `GCC_ARM_PATH` to the global config with 
`mbed config --global ARM_PATH "/usr/bin"

I then import the example program as documented in the readme:
`mbed import https://mbed.org/teams/mbed/code/mbed_blinky/`

Then I try to compile the project with the following command: 
`mbed compile -m LPC1768 -t GCC_ARM`

I get the following error: 
```bash
[ERROR] Attribute 'default_build' not found in target 'LPC1768'
[mbed] ERROR: "python" returned error code 1.
[mbed] ERROR: Command "python -u /home/sille/mbed_blinky/.temp/tools/make.py -t GCC_ARM -m LPC1768 --source . --build ./BUILD/LPC1768/GCC_ARM" in "/home/sille/mbed_blinky"
```

The error is shown on other targets and toolchains as well.

Verbose output:

```bash
mbed compile -m LPC1768 -t GCC_ARM -v
[mbed] Working path "/home/sille/mbed_blinky" (program)
[mbed] Exec "python -u /home/sille/mbed_blinky/.temp/tools/make.py -t GCC_ARM -m LPC1768 --source . --build ./BUILD/LPC1768/GCC_ARM -v" in /home/sille/mbed_blinky
Using targets from /home/sille/mbed_blinky/mbed/9bcdf88f62b0/targets.json
Traceback (most recent call last):
  File "/home/sille/mbed_blinky/.temp/tools/make.py", line 271, in <module>
    name=options.artifact_name)
  File "/home/sille/mbed_blinky/.temp/tools/build_api.py", line 459, in build_project
    extra_verbose=extra_verbose, config=config)
  File "/home/sille/mbed_blinky/.temp/tools/build_api.py", line 320, in prepare_toolchain
    extra_verbose=extra_verbose)
  File "/home/sille/mbed_blinky/.temp/tools/toolchains/gcc.py", line 280, in __init__
    elif target.default_build == "standard":
  File "/home/sille/mbed_blinky/.temp/tools/targets.py", line 225, in __getattr__
    result = self.__getattr_helper(attrname)
  File "/home/sille/mbed_blinky/.temp/tools/targets.py", line 212, in __getattr_helper
    % (attrname, self.name))
AttributeError: Attribute 'default_build' not found in target 'LPC1768'
[mbed] ERROR: "python" returned error code 1.
[mbed] ERROR: Command "python -u /home/sille/mbed_blinky/.temp/tools/make.py -t GCC_ARM -m LPC1768 --source . --build ./BUILD/LPC1768/GCC_ARM -v" in "/home/sille/mbed_blinky"
---
```

andcor02, 2016-11-11:
I would suggest

```
For now, open the mbed.bld file in text editor and replace everything with:

http://mbed.org/users/mbed_official/code/mbed/builds/25aea2a3f4e3
Then use 'mbed deploy' to update libraries to this revision.

```
@screamerbg 

I can confirm this issue, this is due to revision mbed 2.0 9bcdf88f62b0 onwards. Works prior.

Regards,

Andrea

sillevl, 2016-11-11:
Thank You, I can confirm that this works.

jensstruemper, 2016-11-16:
Hi, 
is there are different workaround available? Reason: I can only get BLE working with the latest mbed 2 release 129 for all NRF51 targets (online ide). It would compile with release 127, advertising is working,  but I just can't get a successfull connection(Gatt Error).  I tested BLE HRM & UartLoopback against NRF51_DK and HRM1017 targets in booth Online IDE and mbed-cli. 

Cheers,
Jens

andcor02, 2016-11-17:
@screamerbg @theotherjimmy 

This is still an issue and really needs to be addressed.

theotherjimmy, 2016-11-17:
@jensstruemper Right now you can replace the line (280) of toolchains/gcc.py:
```python
elif target.default_build == "standard":
    use_nano = False
elif target.default_build == "small":
    use_nano = True
```
with 
```python
elif getattr(target, "default_build", None) == "standard" or getattr(target, "default_lib", None) == "std":
    use_nano = False
elif getattr(target, "default_build", None) == "small" or getattr(target, "default_lib", None) == "small:
    use_nano = True
```

you get this error instead:
```
[ERROR] Feature 'LWIP' is not a supported features
```

which you can get rid of by removing lines 675-678 of tools/config.py . That should be this:
```python
        for feature in features:
            if feature not in self.__allowed_features:
                raise ConfigException(
                    "Feature '%s' is not a supported features" % feature)
```

Then it works.

PLEASE NOTE: This is clearly a hack. I'm working on updating the mbed-sdk-tools repo to use the most recent released version of the tools from mbed-os.

jensstruemper, 2016-11-18:
@theotherjimmy Thanks! Working fine. Happy to apply this until mbed-sdk-tools are updated.

abouillot, 2017-01-14:
I have the same issue with NUCLEO_L152RE based  program. I'm using mbed-dev, so a mbed-dev.bld and not mbed.bld. The 25aea2a3f4e3 revision doesn't exist for this variant. What would be the proper workaround?

theotherjimmy, 2017-01-17:
@abouillot This issue will persist until https://developer.mbed.org/users/mbed_official/code/mbed-sdk-tools/pull-request/1 is merged, and you update your tools for your project. In the mean time, you could clone the branch mentioned in that PR as your tools, or use the workaround I mentioned above.

theotherjimmy, 2017-02-27:
That PR has been mainlined. Please run `mbed update`. Closing.

sillevl, 2017-02-28:
Did a small test and I could not reproduce the problem with revision `ad3be0349dc5`. I will close this issue

closed 2017-02-28 by sillevl