ARMmbed/mbed-cli #392: Documentation improvement: Setting toolchain in osx for armgcc and other stuff

shayo, 2016-11-12:
This is a documentation issue.
Can you please explain how do I set the toolchain in osx (you only mentioned for windows).
I'm using arm-none-eabi-gcc

Also, what is the proper target name for nrf51dk development board?

Thank you!

bridadan, 2016-11-14:
@shayo The methods described in the README should work on OSX as well as Windows. We used Windows paths in the README but it should work with any type of path.

Also, if `arm-none-eabi-gcc` is in your shell's PATH, then it should be automatically detected with no need for any configuration!

> Also, what is the proper target name for nrf51dk development board?

`NRF51_DK` :)

shayo, 2016-11-14:
Gotcha. I managed to run the compile. Now I get a crash. I'll open another issue.

closed 2016-11-14 by shayo