ARMmbed/mbed-cli #393: Documentation request - soft device / bootloader ?

shayo, 2016-11-15:
When I compile my project with mbed cli, I get a pretty large hex file (NRF51_DK).
I assume it already has the softdevice? Can you please provide more details? which softdevice is used? from which SDK? does the hex also contain a bootloader? which bootloader? how can I compile and get only the app hex, in case I want to use a different boot loader?

bridadan, 2016-11-15:
cc @pan-

shayo, 2016-11-16:
OK. After digging in the files, I think I realized what is going on.
targets.json describes all compiled configurations.
NRF51_DK_BOOT - includes the bootloader and soft device
NRF51_DK_OTA - compiles only the app.

Can you please confirm?

bridadan, 2016-11-16:
I believe this is the intended behavior, yes. Does that sounds right @0xc0170 ?

shayo, 2016-11-16:
Can you also please confirm whether the default bootloader is dual bank or single bank?

pan-, 2016-12-12:
@shayo This is getting reworked: https://github.com/ARMmbed/mbed-os/pull/3376 . 
Documentation will be added on platform pages.

nvlsianpu, 2016-12-13:
> I believe this is the intended behavior
Yes It's:
NRF5x_DK = Application + SoftDevice
NRF5x_DK_BOOT = Application + SoftDevice + Bootloader
NRF5x_DK_OTA = Application

nvlsianpu, 2016-12-16:
I believe this https://github.com/ARMmbed/mbed-os/pull/3453 will resolve this.
Some base clarification here https://developer.mbed.org/platforms/Nordic-nRF52-DK/#ble-stack (and for nrf51 dk & dongle as well).

theotherjimmy, 2017-11-28:
The PRs mentioned in this issue are long merged. Closing.

closed 2017-11-28 by theotherjimmy