ARMmbed/mbed-cli #394: Callback.h crash on #include <new>

shayo, 2016-11-16:
I'm getting this annoying compilation error, which complains about a missing h file.
The crash is at ./mbed-os/platform/Callback.h:21, which attempts to #inclulde <new>

Any ideas what is going on?
I wasn't able to reproduce this on a small project (i.e., blinky does compile properly). It only happens on my large project.

See below:

Compile [ 89.5%]: nrf_drv_ppi.c
Compile [ 89.8%]: nrf_drv_gpiote.c
Compile [ 90.2%]: nrf_drv_spis.c
Compile [ 90.5%]: nrf_drv_timer.c
Compile [ 90.9%]: bootloader_util.c
Compile [ 91.3%]: dfu_init_template.c
[ERROR] In file included from ./mbed-os/rtos/Thread.h:27:0,
                 from ./mbed-os/rtos/rtos.h:28,
                 from ./mbed-os/mbed.h:22,
                 from ./CRC16/crc16.h:4,
                 from ./mbed-os/targets/TARGET_NORDIC/TARGET_NRF5/sdk/libraries/bootloader_dfu/dfu_init_template.c:75:
./mbed-os/platform/Callback.h:21:15: fatal error: new: No such file or directory
compilation terminated.

[mbed] ERROR: "python" returned error code 1.
[mbed] ERROR: Command "python -u /Users/shayo/mbed/hr/mbed-os/tools/make.py -t GCC_ARM -m NRF51_DK --source . --build ./BUILD/NRF51_DK/GCC_ARM" in "/Users/shayo/mbed/hr"

---

shayo, 2016-11-16:
I think this was somehow related to an existing crc16 file in my project.
Probably not a real issue

closed 2016-11-16 by shayo

pfried, 2017-01-27:
Actually i think this is a real issue, i am having the same problem and i think there is some misconfiguration of the toolchain

For the blinky example this problem does not occur, but if you have files outside of the mbed-os folder this error comes up

SummerSun, 2017-04-13:
Even though the issue was closed already, I think it still could be benefit if anyone else meet the similar errors and be guided here by google search.
It does crash on #include \<new\> when reference mbed c++ from .c file, root cause is still in progress, will update when I found anything.
Back to the issue, quick solution here is to change the .c file to .cpp. Everything works fine. If you would like to keep the file c style, **extern "C" block** would help.