ARMmbed/mbed-cli #395: Supporting floating point numbers in printf / scanf (ARM_GCC)

jensstruemper, 2016-11-16:
Hi, 
currently printf / scanf do not support floating point numbers when compiling with ARM_GCC. This is understandable as floating point support increases the binary size. One the other hand it's often handy for debugging / troubleshooting purposes. 
Could this be made configurable, perhaps as a command line switch? Related issues:

ARMmbed/mbed-os#2012

Thanks !

bridadan, 2016-11-16:
@jensstruemper You should be able to support this by modifying/creating a new build profile: https://github.com/ARMmbed/mbed-os/blob/master/tools/profiles/default.json

You can either use the `default.json` as a template or just modify it directly (though this will make upgrading mbed OS a little more difficult). You should be able to add the necessary flags to enable floating point number in printf/scanf.

@theotherjimmy is this how'd you recommend enabling this?

jensstruemper, 2016-11-16:
@bridadan thanks for the hint. Just can't find default.json nor the profiles folder. It's not part of mbed-cli / mbed-tools or within the project folders. Where should it be located?

bridadan, 2016-11-16:
Oops, sorry my answer was a bit short :)

If you're working in an mbed-os application, it should be located in `mbed-os/tools/profiles/`. If you don't see that directory you may be using an older version of mbed OS. This was added in mbed OS 5.2.0: https://github.com/ARMmbed/mbed-os/pull/2802

jensstruemper, 2016-11-16:
Ah, ok. My installation might be still a bit outdated from the first trial 3 month ago. Actually I only updated mbed-cli to the latest version and then imported one of my mbed os 2 (classic) programs. Compiling works fine do I need to update something else? I was under the impression that the "OS" or "Libraries" are imported when issuing `mbed import bla `.

bridadan, 2016-11-16:
Ah gotcha. The mbed os 2 (classic) programs use a different version of the tools, so I'm not sure about those.

But if you want to use the latest version of mbed OS, you can either start a new project (`mbed new myproject`) or you can update your mbed-os to the latest release. (`cd mbed-os` then `mbed update mbed-os-5.2.2`)

jensstruemper, 2016-11-17:
Ok, know it makes sense to me. When I create a new mbed-os project the profile files are there. Will mbed os 2 build process stay as  is or are you planning to update it to the new mbed-os approach? Also, in the previous version I just appended the necessary flags somewhere in gcc.py (within the .temp directory if I remeber correctly) but the file has changed and it is no longer obvious where this could be done. Any idea?

bridadan, 2016-11-17:
Not sure about the mbed OS 2 build process myself. But as far as what files to modify, you don't need to modify the python file anymore, you can use the build profile json files I mentioned above. You can also create a custom profile and use the `--profile <path to file>` option when using mbed-cli. Otherwise it will just use `default.json`.

theotherjimmy, 2016-11-17:
@bridadan That would be the case if https://developer.mbed.org/users/mbed_official/code/mbed-sdk-tools was the same version as the mbed-os tools, but that is simply not the case. It still uses the `-o` options.

jensstruemper, 2016-11-17:
How would `-o`work? Besides 3 options which seem to be unrelated I cant find any information on how to use it. I think you initially wanted to pass compiler flags. https://github.com/ARMmbed/mbed-os/pull/2431. If I try something like `mbed compile -m HRM1017 -t GCC_ARM --ldflags=-Wl,-u,_printf_float` there are no errors but it has no effect.

jensstruemper, 2016-11-18:
gcc.py is after me. I just applied another patch and realised that I could simply add `"-u _printf_float", "-u _scanf_float"` to the `DEFAULT_FLAGS`. How could I oversee this. Thanks for your support.

jensstruemper, 2016-12-07:
Started to migrate one of my old v2 projects to mbed-os. I modified the` debug.json` profile and appended `"-u _printf_float", "-u _scanf_float"` to the ld flags but it has no effect. 

Any idea? 

`mbed compile -v -t GCC_ARM -m HRM1017 --profile mbed-os/tools/profiles/debug.json
`
```
"GCC_ARM": {
        "common": ["-c", "-Wall", "-Wextra",
                   "-Wno-unused-parameter", "-Wno-missing-field-initializers",
                   "-fmessage-length=0", "-fno-exceptions", "-fno-builtin",
                   "-ffunction-sections", "-fdata-sections", "-funsigned-char",
                   "-MMD", "-fno-delete-null-pointer-checks",
                   "-fomit-frame-pointer", "-O0", "-g"],
        "asm": ["-x", "assembler-with-cpp"],
        "c": ["-std=gnu99"],
        "cxx": ["-std=gnu++98", "-fno-rtti", "-Wvla"],
        "ld": ["-Wl,--gc-sections", "-Wl,--wrap,main", "-Wl,--wrap,_malloc_r",
               "-Wl,--wrap,_free_r", "-Wl,--wrap,_realloc_r",
               "-Wl,--wrap,_calloc_r", "-Wl,--wrap,exit", "-Wl,--wrap,atexit", "-u _printf_float", "-u _scanf_float"]
    }
```

jensstruemper, 2016-12-07:
ah, mistake. I didn't change anything in the code so a compile actually didn't happen and consequently floating point support wasn't added.

closed 2016-12-07 by jensstruemper