ARMmbed/mbed-cli #396: Installing mbed-cli fails to install prettytable

enrcor01, 2016-11-21:
This is same a #245. It was already closed but I am facing the same issue
I install the latest mbed-cli:

```
sudo pip install mbed-cli
The directory '/home/enrique/.cache/pip/http' or its parent directory is not owned by the current user and the cache has been disabled. Please check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
The directory '/home/enrique/.cache/pip' or its parent directory is not owned by the current user and caching wheels has been disabled. check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
Collecting mbed-cli
  Downloading mbed-cli-0.9.10.zip (48kB)
    100% |████████████████████████████████| 51kB 383kB/s 
Installing collected packages: mbed-cli
  Running setup.py install for mbed-cli ... done
Successfully installed mbed-cli-0.9.10
```

And then when I try to compile my application I get

```
sudo mbed compile -vvv -t GCC_ARM -m K64F
[mbed] Working path "/home/enrique/development/workspaces/app_engineers/mbed-cloud-client-example" (program)
[mbed] Exec "python -u /home/enrique/development/workspaces/app_engineers/mbed-cloud-client-example/mbed-os/tools/make.py -t GCC_ARM -m K64F --source . --build ./BUILD/K64F/GCC_ARM -v" in /home/enrique/development/workspaces/app_engineers/mbed-cloud-client-example
Traceback (most recent call last):
  File "/home/enrique/development/workspaces/app_engineers/mbed-cloud-client-example/mbed-os/tools/make.py", line 45, in <module>
    from tools.options import get_default_options_parser
  File "/home/enrique/development/workspaces/app_engineers/mbed-cloud-client-example/mbed-os/tools/options.py", line 21, in <module>
    from tools.toolchains import TOOLCHAINS
  File "/home/enrique/development/workspaces/app_engineers/mbed-cloud-client-example/mbed-os/tools/toolchains/__init__.py", line 36, in <module>
    from tools.memap import MemapParser
  File "/home/enrique/development/workspaces/app_engineers/mbed-cloud-client-example/mbed-os/tools/memap.py", line 11, in <module>
    from prettytable import PrettyTable
ImportError: No module named prettytable
[mbed] ERROR: "python" returned error code 1.
[mbed] ERROR: Command "python -u /home/enrique/development/workspaces/app_engineers/mbed-cloud-client-example/mbed-os/tools/make.py -t GCC_ARM -m K64F --source . --build ./BUILD/K64F/GCC_ARM -v" in "/home/enrique/development/workspaces/app_engineers/mbed-cloud-client-example"
```

enrcor01, 2016-11-21:
I already tried to install prettytable

```
enrique@enrique-VirtualBox ~/development/workspaces/app_engineers/mbed-cloud-client-example $ pip install prettytable
Requirement already satisfied: prettytable in /usr/local/lib/python2.7/dist-packages
```

screamerbg, 2016-11-21:
@enrcor01 can you please try `mbed new example_program` and let me know what happens?

screamerbg, 2016-11-25:
bump @enrcor01

screamerbg, 2017-01-04:
bump2 @enrcor01

jacobrosenthal, 2017-01-12:
For me as well. 
```
Jacobs-MacBook-Air:mbed-os-example-ble jacobrosenthal$ mbed new example_program
[mbed] Creating new program "example_program" (git)
[mbed] Adding library "mbed-os" from "https://github.com/ARMmbed/mbed-os" at branch latest
[mbed] Updating reference "mbed-os" -> "https://github.com/ARMmbed/mbed-os/#2885c1b41e63158cb6faf5f107cd821ae06ef26c"
```
I had to
```
brew uninstall python
sudo easy_install pip
sudo pip install -r mbed-os/requirements.txt
```

bridadan, 2017-01-12:
@jacobrosenthal Thanks for sharing your experience! Was your problem having two installations of python? And the `brew` installation's packages were superseding your system installation?

jacobrosenthal, 2017-01-12:
Presumably something like that. 

I could run `python` and `from prettytable import PrettyTable` would error, but if I ran `python2` `from prettytable import PrettyTable` would succeed

Not really sure how pathing works and new osx comes with python ~2.7 so just decided to kill brew's python entirely.

bridadan, 2017-01-12:
Ahh yeah, that'd probably do it! I'm not really sure how we can avoid this problem since it will pretty variable between different users' environments.

screamerbg, 2017-07-05:
@enrcor01 Could you please recheck with mbed CLI 1.2.0 and update this issue?

closed 2017-12-02 by screamerbg