ARMmbed/mbed-cli #398: mbed-cli config root fails if main is inside source dir

mazimkhan, 2016-11-22:
mbed-cli config root fails if main is inside source dir.

```
0.14s$ mbed config root .
[mbed] ERROR: Could not find mbed program in current path "/home/travis/build/mazimkhan/mbed-os-example-uvisor-thread".
[mbed] ERROR: Change the current directory to a valid mbed program or use the '--global' option to set global configuration.
---
```

screamerbg, 2016-11-22:
Could you please try 'mbed new.  - - create-only' and let me know if that works?

screamerbg, 2016-11-25:
cc @mazimkhan

mazimkhan, 2016-11-25:
It gives me following output:
```
C:\Users\azikha01\Documents\GitHub\mazimkhan\uvisor-tests\autogen\k64f_gcc_arm_r_1_1_2>mbed new . --create-only
[mbed] WARNING: Directory "C:\Users\azikha01\Documents\GitHub\mazimkhan\uvisor-tests\autogen\k64f_gcc_arm_r_1_1_2" is not empty.
---
[mbed] Creating new program "k64f_gcc_arm_r_1_1_2" (git)

C:\Users\azikha01\Documents\GitHub\mazimkhan\uvisor-tests\autogen\k64f_gcc_arm_r_1_1_2>
```

teetak01, 2016-11-28:
I have the same issue. This https://github.com/ARMmbed/mbed-cli/pull/397 fixes the problem.

screamerbg, 2017-07-05:
This is now part of the mbed CLI 1.2 release. Please update, recheck this and update the issue.

closed 2017-12-02 by screamerbg