ARMmbed/mbed-cli #400: Sync overwrites protocol

betzw, 2016-11-28:
Command `mbed sync` seems to always change the protocol to `https` and doe not maintain the original protocol which in my case could be `ssh`.

betzw, 2016-12-20:
@screamerbg

screamerbg, 2017-01-04:
@betzw Repository locations/URLs are stored in the .lib files as `https` URLs for consistency across the different protocols. Regardless of the format in the .lib files, the protocol can be changed using the --protocol option, e.g. `--protocol ssh`.

Why `https` is the default protocol in mbed CLI? 3 reasons:
* Availability - `https` is widely available and de-facto standard for nowadays web. `ssh` being primarily associated a remote command shell, is often seen as a possible security compromise and unfortunately filtered.
* IT policy - While `https` is widely available in corporate environments, `ssh` isn't. In fact some companies not just block `ssh`, but also monitor any activity. Imagine that you're working in one of these companies and the mbed CLI behavior is to use the whatever protocol is stored in the .lib file. While you're importing a program to try something, you might end up attempting an `ssh` connection to `s3cr3t.jpl.nasa.gov` because that's what someone put in one of the .lib files!
* Ease of use - For someone starting with mbed CLI, a .lib file containing `https://github.com/ARMmbed/mbed-client/` would make much more sense than `git@github.com:ARMmbed/mbed-client.git`. Opening the `https` link will give them the home page of the repository.

betzw, 2017-01-09:
What should I do in a mixed situation, i.e. some .lib files with `https`, some with `ssh`?

bridadan, 2017-01-09:
@betzw to clarify, do you mean that some libraries you wish to clone over `https` and some you wish to clone over `ssh`?

betzw, 2017-01-10:
Yes! 

This is my typical development situation:
- most libs coming from GitHub/ARMmbed
- some libs are still in internal development and not yet published and therefore on an internal Git server supporting only ssh.

Furthermore my problem is less about initial cloning but more about syncing/updating and (internal) deploying.

bridadan, 2017-01-10:
Ah, thanks for clarifying! I don't believe this is a use case we currently support (as you've supported with your comments). Could you not setup SSH for GitHub though? It sounds like you already have SSH keys created for your local server. That way you don't have to switch between authentication methods each time.

bridadan, 2017-01-10:
Looks like this is pretty much the same issue as #418

betzw, 2017-01-11:
Yes it's very similar to issue #418!
The problem with `ssh` & GitHub is that I am behind a damned comapny firewall & our IT department won't let pass `ssh` neither in nor out ...

bridadan, 2017-01-11:
@betzw ah, good point! Seems like a valid use case to me (if inconvenient :smile:). Do you mind if we continue the discussion on #418?

betzw, 2017-01-12:
It's for sure inconvenient :wink:
Ok, with continuing discussion on #418!

Should I close this issue?

bridadan, 2017-01-12:
@betzw I'll take care of it!

closed 2017-01-12 by bridadan